FreeCAD/FreeCAD #14862: Changing document unit system does not trigger 'is-modified' flag for 1st document

Roy-043, 2024-06-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Create 3 new documents.
2. Via the Property View change the unit system in each.
3. Result: 1st document is not marked as changed.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37841 (Git)
Build type: Release
Branch: main
Hash: 5f761c758e2a0164b00a7183b9c0cbe32688e123
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2025-06-19:
Related: the right most widget (showing the Unit System) does not update if the Unit System of the 1st document is changed.

maxwxyz, 2026-11-07:
There seems to be a fix in AstoCAD: https://github.com/AstoCAD/FreeCAD/commit/495cd3b2f5f0c93e883b49c563cebbc2a2bea948

The first document gets the missing relabel notification, so changing its document properties can mark it modified too.