FreeCAD/FreeCAD #14865: Spreadsheet - Incorrect display of result when using exponents in formula

dptsolutions, 2024-06-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

See screenshot of issue. I have a cell with alias `round_clip_opening_width' containing the value 2.8mm. I am using that in the formula `(round_clip_opening_width / 2) ^ 2`. It calculates the result correctly, but displays the result as 1.96mm^2, but I would expect it to simply be 1.96mm. These are not square millimeters.

[image]

### Full version info

```shell
OS: Ubuntu 24.04 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37841 (Git) AppImage
Build type: Release
Branch: main
Hash: 5f761c758e2a0164b00a7183b9c0cbe32688e123
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Spreadsheet

### Anything else?

[Wire Wrap Command Strip Mount Attempt 2.FCStd.zip](https://github.com/user-attachments/files/15942016/Wire.Wrap.Command.Strip.Mount.Attempt.2.FCStd.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-23:
Related: https://github.com/FreeCAD/FreeCAD/issues/13534

pitch-circle, 2024-06-27:
This is not a bug. It is a feature. Most quantities in FC have a unit associated with them. Your clip opening width is a length and it has units of length. Hence, when you square it, the result has units of length^2.

Consider this: you have a square. By definition, its length and width are identical. And you want to compute its area. So you do length^2. By your reasoning, you are asking for an area to be shown in units of length^1.

You now have a cube and want to compute its volume. So you do length^3. What units do you want there?

The solution to your problem is shown in the documentation for the Spreadsheet WB, in the Units sub-section (https://wiki.freecad.org/Spreadsheet_Workbench#Units). Please read the very last line of that section.

Then please consider withdrawing the "incorrect" part of your issue or reformulate the issue into a request for the behaviour to be changed.

The issue flagged above as related is not related and that connection should be undone - the other one has at its heart the fact that shape.volume is a unitless quantity.

To solve your issue, use (round_clip_opening_width / 2mm) ^ 2 . This will give a unitless result. If you want the result expressed as a length, use (round_clip_opening_width / 2mm) ^ 2 * 1mm . I agree it is not very intuitive, but having united quantities is part of the 20+ year old DNA of FC and we are stuck with it.

xtemp09, 2024-09-04:
What is the purpose of keeping this issue open? What developers are planning on doing with this?

If I multiply length by length, I get area.

dptsolutions, 2024-09-05:
It's open because I frankly forgot about it. But I guess since there's a workaround, and it's due to a design decision that's deep in the bowels of the project, there's nothing to be done.

closed 2024-09-05 by dptsolutions