FreeCAD/FreeCAD #14866: BIM: Error - as the model becomes more complex, it becomes slower and slower

gianlucafiore, 2024-06-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

What I was able to detect is that by making a model more complex with information in bim, either by setting levels, ordering, details, section planes, etc. The model becomes worse and worse in manageability. But the strangest thing is that by deleting things that slow down the model, it does not return to normal in terms of processing speed or recalculation

[Test file](https://drive.google.com/file/d/1niGEQ2GpDB_2XavfyG6aC4Jb-HlG2zEH/view?usp=drive_link)

[image]

The first is the most complex of the 3 models, which cannot even move a window or door since everything fails, and takes horrors to recalculate.

the second model, was simply created, and placed within Level 3, I began to perform tests of manipulating the model and it automatically became slow by the mere fact of staying inside the building and level
I then removed the model from there, but it continued to be slow, never to be easy to manipulate again

and the third is simply a model created from scratch without placing it in levels or cutting planes beforehand. It is much easier to manipulate

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37841 (Git)
Build type: Release
Branch: main
Hash: 5f761c758e2a0164b00a7183b9c0cbe32688e123
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Spanish/Argentina (es_AR)
Installed mods: 
  * Dracula 0.0.5
  * OpenDark 2024.3.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-23:
Please provide a demo file and steps how to reproduce. Also are you sure that this is just in the BIM WB?

gianlucafiore, 2024-06-23:
Good morning, the test file was uploaded, I edited the link to make it more visible.

Building 1° that is located in the "BIM Project" was the one I started to work with, following the entire BIM workflow, but after a few hours of work of performing several operations, creating walls, erasing, placing or moving joinery, creating section plans, etc., the recalculation began to be slower and slower.
I thought that perhaps it could be because of the level of complexity that the project was gaining. Then I decided to try if it was my PC that was the problem, I first tried deleting the section planes and lines generated with shape2D View and elements that could perhaps add complexity to the calculation of the work, but I could realize that within the same file I started to create new walls and new levels that by default are dumped into the general root of the project. And these walls were easy to manipulate with carpentry and everything. In the new ones, any minimal intervention is easily carried out
But in the same arhcivo, if I place this simple wall with a window that was easily manipulated, within the BIM project, it immediately blends in with its slowness.
And if you also prefer to remove the wall that you tested, out of the bim project and return to the root of the file. That wall will continue to be slow and difficult to manipulate.

When I talk about difficult to manipulate, I mean that each operation I perform generates waiting times of 20 to 30 seconds of recalculation.

Steps to perform in the test file:

1-Create a set of test walls, with joinery, slabs, beams, etc... (when you manipulate the elements, you will notice that it is done easily and with short recalculation times)
2 - Place everything in a group/folder to make it easier to move in the hierarchy tree.
3 - Place the entire group within the BIM project in one of the Levels (Ex: Level 3)
4- Try to modify that built group, moving a window, or a wall of place. (everything will start to break) and run slow.
5 - Remove the entire Level 3 group, and bring it back to the root of the file as originally created. (Everything will continue to run slow)
6 - try creating new objects, (these will work fine)

maxwxyz, 2024-06-23:
is the performance in the stable 0.21.2 better? What theme are you using in 0.22dev?

gianlucafiore, 2024-06-23:
> Also are you sure that this is just in the BIM WB?

I did not have the possibility to test in another work environment since I only work with BIM. But I see the focus of the problem in that in that test file everything that enters the "Edificio" is no longer easy to manipulate.
I just opened the file. and everything is a little more agile. I suspect it may all come from the "cache" side

maxwxyz, 2024-06-23:
@yorikvanhavre is this BIM specific?

yorikvanhavre, 2024-06-24:
I don't see any real problem with this model... Adding, deleting, modifying stuff seems normally responsive to me (on a Ryzen9 CPU).

There are mechanisms of course that make things slower: Basically everything that is parametric. If an object needs to be recalculate because another has changed, that slows things down. But that's not specific to BIM. 

I always try to minimize these parametric relationships, for ex. make the openings with different objects than the windows, freeze shape2dviews and TD views, etc..

The dependency graph tool can also help there, for example to make sure you don't accidentally made one object depend on another such as your whole building... That happens sometimes...

maxwxyz, 2024-06-25:
could this be closed then?

yorikvanhavre, 2024-06-25:
I guess yes, unless @gianlucafiore thinks there is more to it?

closed 2024-06-25 by maxwxyz