FreeCAD/FreeCAD #14869: macOS: Cannot select text with Shift+Arrow in tree view, rotates view instead

maxwxyz, 2024-06-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

From: https://forum.freecad.org/viewtopic.php?p=767281
Trying to select characters with Shift + Left or Right Arrow key on macOS in the tree view does not select the text but rotates the view. I cannot reproduce this on Windows.

Steps to reproduce:
1. create a new object
2. right click "Rename"
3. try to select with keyboard shurtcuts (Shift + Left Arrow)

[image]

Other reported the same issue in the file save dialog but I cannot reproduce this.

### Full version info

```shell
OS: macOS 13.6.7
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37841 (Git)
Build type: Release
Branch: main
Hash: 5f761c758e2a0164b00a7183b9c0cbe32688e123
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * 3DfindIT 1.2.0
  * freecad.gears 1.2.0
  * OpenTheme 2024.6.22
  * CfdOF 1.25.17
  * Defeaturing 1.2.2
  * Manipulator 1.5.7
  * fasteners 0.5.24
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.19
  * Curves 0.6.39
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-07-18:
@maxwxyz is there anyone working on macos related tickets ?

maxwxyz, 2024-07-19:
@luzpaz not sure

luzpaz, 2024-07-21:
@JohnOCFII would you mind reproducing ?

JohnOCFII, 2024-07-21:
> @JohnOCFII would you mind reproducing ?

Yes, I can reproduce this.  Standard macOS behavior when editing a text field and holding SHIFT and arrow left or right should continuously select text.  For some reason, even though I am editing (in this case, attempting to rename a sketch) the keypress is rotating the object in PartDesign.

```
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38261 (Git)
Build type: Release
Branch: main
Hash: 51789ae8a1a0b1ed20317eaedcd190baac734e3d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.7.3
```
[image]

JohnOCFII, 2024-07-22:
Also confirmed same issue with the File export Dialog.  (same FreeCAD version as earlier report today 38261)
[image]

ryanthara, 2025-01-31:
I can confirm this behavior in the file dialog with version 

`OS: macOS 15.2
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: C/Default (C)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
`

luzpaz, 2025-01-31:
Anyone interested in spearheading this ?

abluescarab, 2025-09-09:
This is happening to me on Windows as well.

timpieces, 2025-12-14:
Ok so from my investigation:
- The issue I was mentioning with Cmd+a and this issue do indeed seem to be the same thing.
- Online documentation for this issue seems scarce, but it seems that MacOS has slightly different behaviour for application-level hotkeys. The TLDR seems to be that if you register a shortcut for something that you put in the application menu bar in QT, then that shortcut will override anything. This doesn't seem to be true of other platforms.

I've looked for a lot of workarounds and there's not many good ones. It seems like there are two potential approaches:
- address every possible widget (e.g. the widget to edit object label) and make it specifically override these menu-bar level shortcuts
- split these global shortcuts in two such that:
    - they are added to the menu bar with no shortcut, and
    - they are added to a locally scoped widget with their shortcut
The second of these approaches would have the effect that the shortcut would no longer work on other platforms when the window isn't focused (e.g. if you have clicked on the tree view then shift+right would no longer rotate the 3d view).

I will probably try to look a little bit further, but it may be hard to fix this functionality on mac without having impacts on other platforms, and it may not be as straightforward of a fix as I'd have hoped.
I was surprised that more applications don't have these issues, and I'd like to look into that further, but I think it's rare to add so many menu bar shortcuts with these types of shortcuts

closed 2025-12-23 by chennes