FreeCAD/FreeCAD #14900: FreeCAD requires downloading untrusted code

Jookia, 2024-06-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently it is impossible to create a FreeCAD source tree that contains only authenticated code: That is, code that is known to be the right versions. For instance:

- The FreeCAD repo does not sign commits, tags or releases
- The pre-commit hooks require arbitrary versions of code
- Git submodules are used which can download arbitrary versions of code

It's unsettling as there is no way to verify the code needed to build or develop FreeCAD is actual code from developers. Someone with access to the pre-commit hook repositories or submodules could push changes that include malicious code.

Fixing this isn't too hard:

1. Pin pre-commit modules by hash instead of branch
2. Somehow pin gitmodules by hash or switch to git subtrees
3. Sign releases and commits using a SSH or GPG key

### Full version info

```shell
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37842 (Git)
Build type: Release
Branch: jookia_testing
Hash: 0f9dcefb3f8a99969117ff60167e6a8fdb26706d
Python 3.12.3, Qt 6.7.1, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.4.67

(Note: Upstream is f294a5efc546eda9f0d6d1b85603544f09053cbe)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Jookia, 2024-06-23:
Uh, I've changed the code to avoid the scare factor. To be clear: This is NOT a user security issue. It is an issue for developers.

Jookia, 2024-06-24:
#14902 can help with this, it's the most immediate problem for me personally.

yorikvanhavre, 2024-06-24:
We could enforce some of these things indeed... Pinning the pre-commit hash seems like a good thing. Pinning sub-modules too, I think it's a common practice. I'm actually surprised it's not the case? I thought git modules by default don't update automatically...

Regarding signing commits: What's the advantage of that, regarding code authenticity and reproducibility? It's certainly enforceable, but I'm worried about the effects on small contributors. Many people who just come to fix a small issues will no longer do it if we add barriers.

Jookia, 2024-06-24:
Hi there.

Git submodules can't be pinned using native Git features, you can only specify a branch or tag, not a commit hash, and you can't specify a command to verify the branch or tag. We would need to add some extra code for checking the submodules. On my tree I've added this script for downloading submodules:

```
#!/bin/sh
git submodule init
git submodule update
git -C src/3rdParty/GSL switch --detach b39e7e4b0987859f5b19ff7686b149c916588658
git -C src/3rdParty/OndselSolver switch --detach 04584185b013887be92ab5354321ce83257c8945
git -C tests/lib switch --detach f8d7d77c06936315286eb55f8de22cd23c188571
```

Regarding signing commits, I think I was a bit unclear: I don't mean PRs or every commit or contribution, just release commits and tags, and maybe even binaries distributed with the releases.

The threat model here is to make sure developers and packagers are able to verify they have the correct source code regardless of the source it is downloaded from. This allows trusting mirrors of the code and release files a lot more, and it avoids the security hazard of what a compromised developer GitHub account can do. There's two ways to do solve this issue: Verify and pin things, or sign things. Signing requires having to verify one thing (the keys) instead of multiple (every commit they download or package.)

Looking at things from that angle, pinning code for dependencies allows FreeCAD developers to vouch for it and ensure the wrong versions aren't being downloaded. Providing signed commits or tags means people don't have to manually verify and pin FreeCAD commits, they just have to check whatever release they download against a key they've verified once.

There's not much benefit for end users as they aren't going to verify builds and usually just download it from a trusted source, but there's a lot of benefit for packagers and developers who want to make sure they aren't getting malicious code. This would also be very useful for an auto update feature if FreeCAD has this.

This also doesn't solve the addon manager being the biggest malware vector for FreeCAD users in practice, but asking addon developers to sign their code and ensure their third party dependencies are pinned or signed is likely the wrong approach for solving that issue.

Personally I'm mainly worried about some GitHub account compromise injecting malware in to FreeCAD code or dependencies. Signing and pinning would solve this issue for me and probably others.

NomAnor, 2024-06-24:
Git submodules are always pinned. A submodule is a reference to a commit in the repository referenced in `.gitmodules`.
If a submodule should be updated a developer has to do a `git checkout <new-commit>` inside the submodule folder and then needs to add that change in the parent repository and commit it. That's why you sometimes have to run `git submodule update` to get the current commits of the submodules after a `git pull` of the parent repository.

Jookia, 2024-06-24:
Git submodules are not pinned, they specify branches not commits. There is no way to specify that a git submodule uses a commit.

NomAnor, 2024-06-24:
```
git submodule status
-b39e7e4b0987859f5b19ff7686b149c916588658 src/3rdParty/GSL
-04584185b013887be92ab5354321ce83257c8945 src/3rdParty/OndselSolver
-f8d7d77c06936315286eb55f8de22cd23c188571 tests/lib
```
These are the current commits that are used for the submodules. If you clone the FreeCAD repository and do `git submodule update --init` you will get those commits and not the HEAD commit of the main branches. The only way to change these commits hashes is to change them in a commit to the FreeCAD repository.
If for example OndselSolver gets new commits in their repository, they are not automatically in the FreeCAD repository. A commit is needed (e.g. https://github.com/FreeCAD/FreeCAD/commit/a0683cce88651ec021a7c732a9b89050c6a46e69).

From https://git-scm.com/docs/gitsubmodules

> [...] the superproject tracks the submodule via a gitlink entry in the tree at path/to/bar and an entry in its .gitmodules file (see [gitmodules[5]](https://git-scm.com/docs/gitmodules)) of the form submodule.foo.path = path/to/bar.
> The gitlink entry contains the object name of the commit that the superproject expects the submodule’s working directory to be at.

Maybe I misunderstand what you mean by pinned?

Jookia, 2024-06-24:
Oh! You're right, I was wrong. That's good news. :)

yorikvanhavre, 2024-06-25:
Ok I see your point. And indeed it seems sound. So basically, since the gitmodules questionis solved,:

- Who makes the release tags need to sign it (easy I guess)
- Who makes the release commit needs to sign it (I  think there is not necessarily a release commit. Releases are done from a tag. So maybe we need to enforce it, like always do a cosmetic commit before tagging?)
- pre-commit needs to be pinned (need to see how)

Jookia, 2024-06-25:
My PR to pin pre-commit is merged, so at this point it is just about signing commits and releases.

yorikvanhavre, 2024-06-26:
@adrianinsaval I guess that's mainly for you ;) I'll let you close this issue once you saw it

luzpaz, 2024-07-13:
Close ticket ?

adrianinsaval, 2024-07-13:
I don't think we're going to implement commit signing but I can do source package signing. I'll keep this open until I do, which would be when we release 1.0

kadet1090, 2024-08-19:
So this come back on Saturday meeting - I was not there but we discussed it later on the discord and Monday merge meeting briefly. 

Basically current incentive is to softly enforce (encourage) commit signing from at least recurring developers. However while it is good practice it does come at a price: People need to set it up (which may scare of some people as it requires some work) and we loose few abilities like updating branches via rebasing from github UI.

I'm however not sure what benefits will it bring to us. Signing commits is crucial for scenarios where we have trusted developers who can commit directly to the repository with minimal or no control. Which is true for many projects but not for FreeCAD - we have strong PR policy and every commit that is incorporated to FreeCAD must be checked at least by two people and often more on the Monday meeting.  Basically signing commits is useful mostly for cases where we want to confirm that it was really the trusted individual who did the commit.

We also deal with a lot of contributions from newcomers and people who contribute very occasionally - signed commits would give only false sense of security in such cases as they only confirm that it was really this person who did the commit. Person that we do not know anyway so that confirmation does not bring anything useful for us.

My proposal is to force signed commits only for the Maintainers and possibly for all the members of FreeCAD github org - that should give us the most benefits and reduce the costs.

Pinging @FreeCAD/developers and @FreeCAD/maintainers for comment.