FreeCAD/FreeCAD #14901: Subtractive helix sometimes adds intead of subtracting

johnsonm, 2024-06-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a part from which I'm trying to cut part of a helical thread, and sometimes the subtractive helix adds instead of subtracting.

I set the Boolean to tip and show the sketch:

[image]

Then set the SubtractiveHelix to tip — and it's adding instead:

[image]

Sometimes editing the helix and changing it to reversed, then back, fixes it. Sometimes it doesn't. Sometimes it causes the entire part to disappear.

The sketch passes all of "Validate Sketch" so it shouldn't be problems with a wire not being closed.

[DremelNut.zip](https://github.com/user-attachments/files/15946657/DremelNut.zip)

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37859 (Git)
Build type: Unknown
Branch: main
Hash: 6fd73487c7ea2c66e42395355de916f718320c9a
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.4
  * Curves 0.6.13
  * sheetmetal 0.3.0
  * ExplodedAssembly
  * fasteners 0.4.66
  * MeshRemodel 1.8919.0
  * freecad.gears 1.0.0
  * Pyramids-and-Polyhedrons
  * dxf_library
  * lattice2 1.0.0
  * boltsfc 2022.11.5
  * in3dca-freegrid 2.0.0
  * parts_library
  * Defeaturing 1.2.0
  * A2plus 0.4.60n
  * CurvedShapes 1.0.4
  * Manipulator 1.5.0
  * Pyramids-and-Polyhedrons.backup1696298372.5270553 (Disabled)
  * Assembly3 0.12.2
  * DynamicData 2.62.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

johnsonm, 2024-06-23:
I tried opening it in the LinkStage version. After re-calculating, it was correct there.

[code]
OS: Fedora Linux 40.20240616.0 (Silverblue) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 2024.322.0.14555 (Git shallow) AppImage
Build type: Release
Branch: (HEAD detached at 20240322stable)
Hash: da98001cfe5cc50cb2e9e4c241626d07dd4b06f5
Python 3.11.6, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.4
  * Curves 0.6.13
  * sheetmetal 0.3.0
  * ExplodedAssembly
  * fasteners 0.4.66
  * MeshRemodel 1.8919.0
  * freecad.gears 1.0.0
  * Pyramids-and-Polyhedrons
  * dxf_library
  * lattice2 1.0.0
  * boltsfc 2022.11.5
  * in3dca-freegrid 2.0.0
  * parts_library
  * Defeaturing 1.2.0
  * A2plus 0.4.60n
  * CurvedShapes 1.0.4
  * Manipulator 1.5.0
  * Pyramids-and-Polyhedrons.backup1696298372.5270553 (Disabled)
  * Assembly3 0.12.2
  * DynamicData 2.62.0
[/code]

[image]

FreeCAD 0.21.2 loads the file with errors and cannot recompute it, so I can't use that as a test one way or another.

maxwxyz, 2024-06-24:
@FlachyJoe FYI

maxwxyz, 2024-06-24:
Related: https://github.com/FreeCAD/FreeCAD/issues/14851

FlachyJoe, 2024-06-24:
The Boolean object has a broken shape so Helix can't do a good job.
[image]

johnsonm, 2024-06-24:
So the bug is that the boolean produced a broken shape?

I had read #14851 before filing this issue and didn't pile onto that one because it was not obvious to me that they were actually related.

FlachyJoe, 2024-06-25:
> So the bug is that the boolean produced a broken shape?

The only thing I can say is that the Helix produce a correct solid
<details><summary>Details</summary>
You can extract the produced shape with<br/>
<code>Part.show(App.ActiveDocument.SubtractiveHelix.AddSubShape)</code><br/>
and then check it with <a href="https://wiki.freecad.org/Part_CheckGeometry ">Part_CheckGeometry </a>
</details> 

so the bug is in the boolean operation with the Helix shape and the previous.
It lets two possibilities:

1. The previous shape is sufficiently  broken to prevent OCC to do the operation
2. The boolean operation is buggy. But As I shown in #14851 the same geometry created in Part fails too so the error is not in the PartDesign algorithm but has to be search in OCC.

luzpaz, 2024-12-09:
Can anyone test on latest dev with OCC 7.8.1 ?

tarman3, 2024-12-09:
FreeCAD freezes after opening file `DremelNut.FCStd`

<details>
  <summary>FreeCAD 1.0.0.39109 (Git) makepkg, Arch Linux (KDE/plasma/wayland)</summary>

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.0; Silk 0.1.5; Curves 0.6.51
```
</details>

FlachyJoe, 2024-12-09:
It opens here (AngularDeviation of the Body is too small so it takes some time to be meshed and displayed) but the shape depends on an external workbench I don't have so I'm unable to recompute.

[image]

```
OS: Debian GNU/Linux trixie/sid (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.39516 +12 (Git)
Build type: Release
Branch: dev
Hash: 562e4c00ea4e0d8b4bee3b92261f99acf9bd31fe
Python 3.12.7, Qt 5.15.15, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.2.dev
Locale: English/United States (en_US) [ OS: French/France (fr_FR) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * fcgear 1.0.0
  * Render 2022.1.0
  * fcpd-workbench 0.3.0
  * Trails 2022.1.0
  * lattice2 1.0.0
  * DynamicData 2.68.0
  * MeshRemodel 1.8919.0
  * SelectorToolbar
  * CfdOF 1.27.11
  * CurvedShapes 1.0.12
  * SearchBar 1.0.2
  * Silk 0.1.5
  * Curves 0.6.50
  * fasteners 0.3.41
  * Assembly4 0.11.12
  * Wood-Frame
  * A2plus 0.4.56
  * Plot 2022.4.17
  * mooc-workbench
  * Nodes 0.1.36
  * sheetmetal 0.2.63
```

stefano2734, 2026-09-12:
It should be tested again with occt 8.0.1 and actual 26.3 dev with libpack 3.5.5 to see the actual state of this problem.
Occt 8 was a huge update with significant changes.