FreeCAD/FreeCAD #14903: Padding sketch around clone fails with "Bnd_Box is void"

johnsonm, 2024-06-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the attached file, selecting Sketch003 and selecting PartDesign_Pad leads to a "Bnd_Box is void" error and no shape is created.

This is true whether I select pad to dimension or pad to face.

I tried changing the inner constraint from 22mm to 21mm in case this was a duplicate of #8640 where the problem is exact overlap, but that didn't fix it.

[DremelNut.zip](https://github.com/user-attachments/files/15948333/DremelNut.zip)

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37859 (Git)
Build type: Unknown
Branch: main
Hash: 6fd73487c7ea2c66e42395355de916f718320c9a
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.4
  * Curves 0.6.13
  * sheetmetal 0.3.0
  * ExplodedAssembly
  * fasteners 0.4.66
  * MeshRemodel 1.8919.0
  * freecad.gears 1.0.0
  * Pyramids-and-Polyhedrons
  * dxf_library
  * lattice2 1.0.0
  * boltsfc 2022.11.5
  * in3dca-freegrid 2.0.0
  * parts_library
  * Defeaturing 1.2.0
  * A2plus 0.4.60n
  * CurvedShapes 1.0.4
  * Manipulator 1.5.0
  * Pyramids-and-Polyhedrons.backup1696298372.5270553 (Disabled)
  * Assembly3 0.12.2
  * DynamicData 2.62.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-24:
Could you test it on the stable version if this is a regression? If possible a screenshot would be nice :)

Roy-043, 2024-06-25:
This issue is caused by an invalid shape. See #14901.

johnsonm, 2024-06-25:
Then my question is the same as in #14901 — are both of these bugs really "boolean operation produces invalid shape"?

maxwxyz, 2024-06-25:
should we close this issue then in favor of #14901 ?

johnsonm, 2024-06-25:
I'll assume that the root cause is the same based on @Roy-043 's analysis.

closed 2024-06-25 by johnsonm