FreeCAD/FreeCAD #14904: Clipping view

malutanpetronel, 2024-06-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The clipping view is not displaying the view in it

### Full version info

```shell
0.21.2
Rev 33771
```

### Subproject(s) affected?

None

### Anything else?

[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-24:
Please add:
- full about info
- demo file
- steps to reproduce

If possible please also test with the weekly version: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

maxwxyz, 2024-06-28:
@malutanpetronel until no further info is provided this issue is closed

closed 2024-06-28 by maxwxyz