FreeCAD/FreeCAD #14905: Core: Selection of Part sub-object in tree highlight whole Part (Regression)

PaddleStroke, 2024-06-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If you select one object within a App::Part in the tree, then the whole App::Part highlight:

[image]

It has been introduced recently, I think at last monday's merge meeting.
@chennes any idea which PR could have introduced this?

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-06-29:
Does this still happen for you? For me it doesn't.
It looks like a duplicate of https://github.com/FreeCAD/FreeCAD/issues/14789 that was fixed with https://github.com/FreeCAD/FreeCAD/pull/14813

PaddleStroke, 2024-07-01:
Indeed its fixed now.

closed 2024-07-01 by PaddleStroke