FreeCAD/FreeCAD #14909: ToolBit Library: Unable to create a new Toolbit in the Toolbit Library editor

DropdeadFredd, 2024-06-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Creating a new Toolbit fails after entering the filename.

CAM Workbench -> Toolbit Library editor -> Create Toolbit

Select the Toolbit Shape and then enter the Toolbit filename. The Toolbit file is created but FreeCAD reports an error:

```
11:13:23  C:/Users/<UserName>/AppData/Roaming/FreeCAD/Tool Library/Bit/Large_endmil.fctb
11:13:23  Traceback (most recent call last):
11:13:23    File "D:\CNC\Apps\FreeCAD v0.22\Mod\CAM\Path\Tool\Gui\BitLibrary.py", line 614, in toolBitNew
11:13:23      self.factory.newTool(self.toolModel, fullpath)
11:13:23      ^^^^^^^^^^^^^^^^^^^^
11:13:23  AttributeError: 'ModelFactory' object has no attribute 'newTool'. Did you mean: 'new_tool'?
```

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37841 (Git)
Build type: Release
Branch: main
Hash: 5f761c758e2a0164b00a7183b9c0cbe32688e123
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * A2plus 0.4.65
  * Assembly4 0.50.13
  * fasteners 0.5.24
```

### Subproject(s) affected?

CAM/Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-28:
Can confirm on mac too.
@sliptonic FYI

spanner888, 2024-08-12:
This appears to have been fixed by  Toolcreatebug #14814 

@ DropdeadFredd  @maxwxyz can you confirm?

knipknap, 2025-05-19:
I can confirm that I do not see this bug in 1.0. So it should be closed IMO.

closed 2025-05-19 by maxwxyz