FreeCAD/FreeCAD #14913: FEM: Erase Elements tool should automatically unhide the original mesh when pressing Restore

FEA-eng, 2024-06-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

As explained [here](https://github.com/FreeCAD/FreeCAD/pull/11492#issuecomment-1926034624), pressing _Restore_ to discard the modifications done with the new Erase Elements tool doesn't bring the original mesh back to visibility which may give an impression that something went wrong.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37841 (Git)
Build type: Release
Branch: main
Hash: 5f761c758e2a0164b00a7183b9c0cbe32688e123
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mac-the-bike, 2024-06-26:
yes you are right !!!
if i come to do a second version then i will change this

FEA-eng, 2024-06-26:
> if i come to do a second version then i will change this

Since the feature is already merged, it should be just a matter of a small PR adding this functionality. But I guess it's for the next release cycle anyway.