FreeCAD/FreeCAD #14914: FEM: Erase Elements tool doesn't support individual element selection

FEA-eng, 2024-06-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

As explained [here](https://github.com/FreeCAD/FreeCAD/pull/11492#issuecomment-1926034624), the new Erase Elements tool supports only Poly selection (picking elements inside or outside a polygon). Selection of individual elements would be very useful here for precise "digging" in meshes.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37841 (Git)
Build type: Release
Branch: main
Hash: 5f761c758e2a0164b00a7183b9c0cbe32688e123
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mac-the-bike, 2024-06-26:
This was something I never thought about.
I can see that it is difficult to choose 1 element with the poly selection. 
Elements are chosen to be erased when its CG is inside/outside the poly. The calculation of the CG is only approximate, if the element is a 20 node brick only the 8 corner grids/nodes are used. (grids if you a NASTRAN user)

FEA-eng, 2024-06-26:
@mac-the-bike Yeah, it would be very convenient when taking a closer look at the insides of the meshes:

[image]