FreeCAD/FreeCAD #14919: Preferences / User Parameters - defaults need to be duplicated

kadet1090, 2024-06-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

All settings are implemented by User Parameters - either via our custom `Pref` widgets or by manually reading and writing to user parameters. Furthermore, every time some User Parameter is used user must provide default value which must be duplicated across the codebase:

[image]

This leads to many inconsistencies, visible mostly in the preferences. Furthermore each change forces us to update potentially multiple places which are very hard to find and so can easily lead to inconsistencies for new users that do not have the default value specified.

### Full version info

```shell
0.22dev
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-06-26:
Related issue: #10176

The solution adopted in the Draft WB is to read the defaults from the ui files used for the preferences: #11677

I think the TechDraw WB sets all missing preferences at start up. This allows to define all defaults in a single file. That would be another potential solution. The defaults in the ui files are then just dummy values.

Roy-043, 2024-07-20:
A good example of this issue is the handling of the Tree view column preferences "Hide description" and "Hide internal names". If you delete your CFG and restart you have a  single column in the Tree view. But after opening the Preferences editor and only pressing OK you have 3 column (to display the column headings you have to restart BTW).

The simplest solution would be to check the related checkboxes in the UI file (but that does not fix the heading issue of course).

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38231 (Git)
Build type: Release
Branch: main
Hash: 29889bfd9a7ba0f0496f2e2699d454ea5fb5a05a
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

adrianinsaval, 2024-10-06:
This also affects us at ondsel as a branded version of freecac, with the branding system it is possible to set a cfg template that is used as a base for user.cfg when it doesn't exist so in this way we can easily set defaults that are different than freecad's but unfortunately if the user then manually resets any preference it pulls in freecad's default and not our default. IMO we need a centralized way of setting default values, probably better if addons can add their defaults to it too.

AIRCAP, 2024-10-29:
Something to add - there are "hidden" preferences - used for development/debugging things that the user does not need to know about.
These are preferences initialized in the code, but not present in the ui file for the config editor, since they don't show up there at all, they can however be manually set via Tools->Edit Parameters  or by editing the config file by hand and adding them.

here is an (incomplete) list of hidden preferences

https://wiki.freecad.org/Fine-tuning

chennes, 2024-10-29:
Just a note that I am experimenting with an alternative preferences system in the Addon Manager -- it wasn't originally developed specifically with the problem in mind, but it does address it. Preference defaults are defined in a JSON file at the top level of the module, and a wrapper around the FreeCAD preferences system then provides access to preferences via `get` and `set` methods that do *not* take a "default" parameter. The wrapper pulls the default from the JSON file. It is an error to request a preference not defined in that file. This could be seen as a proof-of-concept for a more broadly applicable preferences system revamp.