FreeCAD/FreeCAD #14920: Dart Theme interface problems

Pendelf, 2024-06-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Different arrows in the construction tree
Headers in customization are not fully visible

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37841 (Git)
Build type: Release
Branch: main
Hash: 5f761c758e2a0164b00a7183b9c0cbe32688e123
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * A2plus.backup1693145886.351258 0.4.60n
  * Assembly3.backup1697878740.3967428 0.12.0
  * Assembly4.backup1697878772.1268227 0.50.3
  * BIM.backup1704622292.3622465 2021.12.0
  * CurvedShapes.backup1704622353.7785401 1.0.5
  * Curves.backup1693145923.4219682 0.6.13
  * Design456.backup1712597574.1363904 0.0.1
  * dodo.backup1712597608.2771313 1.0.1
  * ExplodedAssembly
  * fasteners 0.5.24
  * FEMbyGEN 2.1.0
  * lattice2 1.0.0
  * LCInterlocking
  * Manipulator 1.5.7
  * PieMenu 1.7.0
  * Pyramids-and-Polyhedrons
  * sheetmetal 0.4.20
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2024-06-24:
Wonder if this is caused by qtstyle.
Because I have not seen this issue in windows 10.

Pendelf, 2024-06-25:
It's better on Linux, but it's like the headers in the settings aren't quite in their proper places anyway
[image]
OS: Manjaro Linux (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git) AppImage
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods:
  * Assembly3 0.12.2
  * fasteners 0.5.21
  * ExplodedAssembly
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.14
  * dodo 1.0.1
  * Curves 0.6.36
  * A2plus 0.4.64a
  * LCInterlocking
  * lattice2 1.0.0

MisterMakerNL, 2024-06-25:
Btw in my new pr I set Qtstyle to fusion because to get a good UI it is nice that everyone is using settings.
I also think normal users won't notice any difference so having different styles won't do anything but cause lot's of weird bugs.

Pendelf, 2024-08-06:
The latest version seems to be fine, the headers are in their proper places

[image]

OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38419 (Git)
Build type: Release
Branch: main
Hash: 270bb0122ea15bbc562942b9d0756fabb89362c0
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * A2plus.backup1693145886.351258 0.4.60n
  * Assembly3.backup1697878740.3967428 0.12.0
  * Assembly4.backup1697878772.1268227 0.50.3
  * BIM.backup1704622292.3622465 2021.12.0
  * CurvedShapes.backup1704622353.7785401 1.0.5
  * Curves.backup1693145923.4219682 0.6.13
  * Design456.backup1712597574.1363904 0.0.1
  * dodo.backup1712597608.2771313 1.0.1
  * ExplodedAssembly
  * fasteners 0.5.24
  * FEMbyGEN 2.1.0
  * lattice2 1.0.0
  * LCInterlocking
  * Manipulator 1.5.7
  * OpenTheme 2024.7.3
  * PieMenu 1.7.6
  * Pyramids-and-Polyhedrons
  * sheetmetal 0.4.23

maxwxyz, 2024-08-06:
Is the arrow issue also resolved?

Pendelf, 2024-08-07:
> Is the arrow issue also resolved?

Yeah, everything seems fine

closed 2024-08-07 by maxwxyz