FreeCAD/FreeCAD #14922: Help module: What's This? stopped working

FEA-eng, 2024-06-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

For some reason, [Std WhatsThis](https://wiki.freecad.org/Std_WhatsThis) is not working for me in the newest weekly build on Windows:

```
20:01:00  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\username\AppData\Roaming\FreeCAD\Mod\Help\.\Help.py", line 108, in show
    show_tab(html, baseurl, title, view)
  File "C:\Users\username\AppData\Roaming\FreeCAD\Mod\Help\.\Help.py", line 214, in show_tab
    if get_qtwebwidgets(html, baseurl, title):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\username\AppData\Roaming\FreeCAD\Mod\Help\.\Help.py", line 234, in get_qtwebwidgets
    import WebGui
  File "E:\FreeCAD_weekly-builds-37841-conda-Windows-x86_64-py311\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'ModuleNotFoundError'>: No module named 'WebGui'
20:01:00  QWindowsWindow::setMouseGrabEnabled: Not setting mouse grab for invisible window QWidgetWindow/'QWhatsThatClassWindow'
```

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37841 (Git)
Build type: Release
Branch: main
Hash: 5f761c758e2a0164b00a7183b9c0cbe32688e123
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-06-25:
Update: it's working in the newest weekly on another computer but it uses a web browser even though Display is set to "In a FreeCAD tab". Also, there are those warnings:

`QWindowsWindow::setMouseGrabEnabled: Not setting mouse grab for invisible window QWidgetWindow/'QWhatsThatClassWindow'`

yorikvanhavre, 2024-06-25:
Since the web WB has been removed, FreeCAD has no more browser capabilities. So the output is directed to the desktop browser if you don't have the necessary PySide support.

Do you have a " PySide QtWebEngineWidgets module is not available. Help rendering is done with the system browser" message in your Report window?

FEA-eng, 2024-06-25:
> Since the web WB has been removed, FreeCAD has no more browser capabilities. So the output is directed to the desktop browser if you don't have the necessary PySide support.

Ok. Then the aforementioned preference should be removed or is it working but only when PySide is installed? Is there an easy way to set it up on Windows? I've only found that it should be available with FreeCAD installation: https://wiki.freecad.org/PySide

> Do you have a " PySide QtWebEngineWidgets module is not available. Help rendering is done with the system browser" message in your Report window?

When using "What's This?"? No, I only have the above warning about mouse grab.

yorikvanhavre, 2024-06-25:
> When using "What's This?"? No, I only have the above warning about mouse grab.

Weird... I don't have any mouse grab message and the help page opens. Does it work if you change the Help preferences to open n desktop browser?

FEA-eng, 2024-06-25:
> Weird... I don't have any mouse grab message and the help page opens. Does it work if you change the Help preferences to open n desktop browser?

It opens for me too so it's not that it doesn't work. There's just this strange warning when using the tool.

> I'm not sure it can be installed afterwards in Windows. It should be bundled with the installer... But maybe on windows we'd need to disable these options

I am using weekly builds on Windows for now. Perhaps it will be different when 1.0 package installer is released.

Roy-043, 2024-06-25:
@FEA-eng You need to remove the Help Addon. The error message in the original report shows you still have it installed.

FEA-eng, 2024-06-26:
> @FEA-eng You need to remove the Help Addon. The error message in the original report shows you still have it installed.

The problem is that I also have 0.21.2 installed. But I guess I have to get rid of it on at least one computer and only keep clean 0.22 since there are too many conflicts between those versions.

FEA-eng, 2024-06-28:
I've tried on a computer with cleaned-up FreeCAD config files and there's still that "mouse grab" warning while the wiki page opens in a web browser (despite the "In a FreeCAD tab" setting). So only the errors from the initial report are gone.

yorikvanhavre, 2024-07-01:
We have here the same problem as BIM: The addon should be removed, yet there are people who still run 0.21 together and want the addon to stay installed...

We did a manipulation in BIM to rename the addon, but here it won't work as the Help module is needed by FreeCAD, it can't be renamed... Not sure how to solve this :/

yorikvanhavre, 2024-07-02:
In the meantime I added a workaround for the remaining issue (what's this opens in desktop browser regardless of the pref option) by adding a note to the preferences dialog. Not ideal but at least people will know about it

Not sure how we can fix the Help addon problem, but after 1.0 is out the problem should vanish as we can assume people won't keep 0.21

PR #15115 (2024-07-08): Help: Added explaining note to preferences dialog
Fixes #14922
commit c30385ed (2024-07-08): Help: Added explaining note to preferences dialog (#15115)

* Help: Added explaining note to preferences dialog - fixes #14922

* Help: Fixed typos
closed 2024-07-08 by chennes