FreeCAD/FreeCAD #14923: Fasteners inclusion distort Assembly WB joints LCS position

MiqCG, 2024-06-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When adding a Fastener WB element to an assembly (of the Integrated Assembly WB), the selection of geometries to attach Joint Coordinate System (JCS) becomes impossible to position properly. 
See attached video and files.

I have made 2 test files with different joints and different bodies just in case the body "#110-002 Trapezi superior Roll", that contain mirrored parts, was creating the distortion. Both test files are made with "insert component" tool. I have tried with 0.22 revisions 37543, 37819 and 37841 with the same results.

And have searched in Github with keywords [is:issue is:open fasteners assembly joint ] and FreeCAD Forum with [assembly fasteners joint] with no relevant results.

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37841 (Git) AppImage
Build type: Release
Branch: main
Hash: 5f761c758e2a0164b00a7183b9c0cbe32688e123
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Installed mods: 
  * Movie 2024.2.15
  * sheetmetal 0.4.19
  * FreeCAD-motion-workbench-master
  * dodo 1.0.1
  * PieMenu 1.2.7
  * BillOfMaterials 0.0.8.5
  * fasteners 0.5.24
```

### Subproject(s) affected?

Assembly

### Anything else?

https://github.com/FreeCAD/FreeCAD/assets/18265355/25ad3288-28dc-4b35-a5ea-20d8d5a9e058

[Ass_fasteners.zip](https://github.com/user-attachments/files/15959802/Ass_fasteners.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-25:
I don't understand this issue:
You try to select the JCS reference on the fastener object which is not part of the assembly? --> won't work
You try to select an edge hidden by an object --> Position it somewhere else?

MiqCG, 2024-06-25:
@maxwxyz , please, look again the attached video an observe where the JCS appear before the inclusion of the nut (they appear on the corresponding geometry) and after the inclusion (they appear on other parts of the 3D view, far from the preselected geometry).

In fact, I am not trying to select the nut in any moment, although it would be great being able to do it.

PaddleStroke, 2024-06-27:
This is weird, I'll check the files and feedback

PaddleStroke, 2024-06-27:
Ok I found the issue. The fastener is linking to the Assembly object. ie assembly is in the outlist of fastener.
So RootObjectsIgnoreLinks does not return assembly.
So getGlobalPlacement is failing.

The solution should be to use TreeRootObjects instead of RootObjectsIgnoreLinks. But the problem is that TreeRootObjects  is in GUI while getGlobalPlacement is in APP.

This is going to be a PITA to fix. Perhaps it's time to rewrite getGlobalPlacement so that it does not rely on RootObjects. Which is going to be super PITA.

closed 2024-07-29 by yorikvanhavre