FreeCAD/FreeCAD #14927: [TNP] Adding windows and doors to a wall changes the numbering of vertices/edges/faces

kaiwas, 2024-06-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In this example, the window coordinates use expressions that take values ​​from the coordinates of the wall vertices.
When we insert windows or doors into a wall, the numbering of all constituent elements changes.

To reproduce the problem, simply assign the host to a door or window. An example is shown in the video.

https://github.com/FreeCAD/FreeCAD/assets/32273833/98ced1cd-e70b-48a4-8ee5-5be0d26f8bf2

Test file is attached. Change the extension .txt to .FCStd
[Test_Walls_and_Windows.txt](https://github.com/user-attachments/files/15966044/Test_Walls_and_Windows.txt)

### Full version info

```shell
0.22dev
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-06-25:
Related issue: #14928.

The TNP solution introduced in the current dev cycle (v0.22/v1.0) relies on a shape history. Walls based on wires and sketches do not have such a history. The intermediate shapes used to generate them are not preserved. I am currently not sure how this can be addressed.

yorikvanhavre, 2024-06-25:
Should we introduce a shape history? Not sure if that's already done in python somewhere...

bgbsww, 2024-07-01:
This appears to be a question of how BIM is implemented ( as Roy-043 notes ).  Is there somebody who knows BIM that can take the point on this?  I'm happy to collaborate, but really don't know anything about the internals of this module.

yorikvanhavre, 2024-07-02:
Walls are constructed by taking a base wire coming from another object (Draft line or sketch), then constructing a face by offsetting that wire and joining the ends, then extruding it to form the wall solid. Most BIM objects do something in those lines too (a column takes a base face and extrudes it, a window takes a series of closed wires and extrudes/cuts them, etc.

This is all done in Python, for ex. here is the BIM panel code, I think it's the easiest one to understand how it works, it basically takes a base 2D shape then extrudes it: https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/BIM/ArchPanel.py#L112

I'm not sure what we need to do here. How would we give all this a shape history?

bgbsww, 2024-07-02:
Thanks, that's very helpful.  I think there's a list of things to verify.  Starting at https://github.com/FreeCAD/FreeCAD/blob/2a009589f9a37b61da7437c49dd7f05148299c3e/src/Mod/BIM/ArchPanel.py#L175 presumably the Base.Shape is a TopoShape and potentially has an element map.  Then there are series of calls to Part operations like copy,makeCompound, extrude, makeShape, ArchCommands.getShapeFromMesh, makePolygon, Part.Face.

I think that if all of these operations correctly update the element map then the underlying history mechanisms should apply here and solve the problem at the level, without having to add new code at the BIM level; also, this should apply almost anywhere we are building on Part operations.

So the first question is are all of those operations element aware and bug free?  I was initially most suspicious of getShapeFromMesh, but it devolves into a series or makeSolid, makeShell, and makeCompound calls.

That leads directly to Part/App/AppPartPy.cpp and checking that everything there is correctly calling the makeElementSomething and not the makeSomething calls.

At a glance, at the very least sortEdges doesn't look right.  Anyplace using a TopoDS_Shape and not a TopoShape is suspect.  The missing makeWires is also suspect.

It is worth noting that this file was done before we discovered that the toponaming branch was incomplete and we needed to cross check on LS3.  Does not look like this has been done here.

Summing all that up, I think we need test coverage on the methods here to prove they are all TNP safe, and then to correct any that are not.  That most definitely is in scope for the TNP migration, but we could certainly use some help here.

bgbsww, 2024-07-02:
Oh, I'm making an assumption here.  Changing the numbering of vertices/edges/faces is unavoidable - that's what happens when a model change happens.  The challenge is making sure that anything that refers to those gets correctly updated.  If there is a correct element map then this should happen magically when we go to get those elements.  Unless there's some hidden code ( like there was in assembly ) that is saving off the element numbers independent of the link mechanism.  For assembly Paddlestroke had to switch storage of element names in String properties to storage in XLink properties.

bgbsww, 2024-07-02:
I'd probably instrument up that code with a bunch of print(DocObject.Shape.ElementMapSize) calls - old school, I know.  Then do whatever calls it and look to see where the element maps are and if any of the operations are losing them as a way to narrow in on probable areas with issues.  If it weren't immediately clear the issue, then c++ breakpoint at the problem calls and step in to watch what is happening.

yorikvanhavre, 2024-07-02:
> Oh, I'm making an assumption here. Changing the numbering of vertices/edges/faces is unavoidable - that's what happens when a model change happens.

Indeed. Imagine a pretty common operation in BIM, you have a rectangle-shaped wall (8 verts), and you add a window to it (plus 8 verts for the hole, becomes 16 verts). Over the course of the project, you might then remove the window, add more, etc.

> presumably the Base.Shape is a TopoShape and potentially has an element map. Then there are series of calls to Part operations like copy,makeCompound, extrude, makeShape, ArchCommands.getShapeFromMesh, makePolygon, Part.Face.

> That leads directly to Part/App/AppPartPy.cpp

Yes. Indeed I think we can safely assume most of the issue is there

> Summing all that up, I think we need test coverage on the methods here to prove they are all TNP safe, and then to correct any that are not. That most definitely is in scope for the TNP migration, but we could certainly use some help here.

Tell us what we can do to help!

bgbsww, 2024-07-02:
Either of two paths:  Python test coverage on those Part.makeX calls that checked for reasonable element maps would be a big boon ( and likely pay off in the long run ),  or the shorter running down which Part.X methods specifically in ArchPanel aren't showing TNP awareness.

Both of those are about narrowing down where the bug is for a quick fix.

Or, there's that side-by-side and code transfer in the .cpp file which I'm probably most experienced with, but I won't get to for some days or weeks.  This one is actually one of the easier ones;   Look for any TopoDS_Shapes that should be TopoShapes.

Unfortunately, I need to do this work on Sketcher right now, and that has me occupied.

More than happy to review and comment if someone picks this up - it's the legwork part that occupies the time.

yorikvanhavre, 2024-07-02:
> Either of two paths: Python test coverage on those Part.makeX calls that checked for reasonable element maps would be a big boon ( and likely pay off in the long run ), or the shorter running down which Part.X methods specifically in ArchPanel aren't showing TNP awareness.

Could you give a quick explanation here, what should we be looking for?

bgbsww, 2024-07-02:
My apologies, I forget how much of this is inside knowledge:

The most basic test here is that a Toposhape ( the Part.Shape in python ) has a non empty elementmap  ( Part.Shape.ElementMapSize > 0 ).   Operations that aren't TNP aware will use TopoDS_Shape in the c++ code and that's the raw OCCT object with no element map.  Operations that are TNP aware will use TopoShape which is a ComplexGeoData that has the underlying TopoDS_Shape, but also has an elementmap.

While there were bugs related to what is in the element, those appear to be nearly squashed, so you don't need to care about the contents.  If you're interested, you can look in Part.Shape.ElementMap or Part.Shape.ElementReverseMap (not always the same size due to duplicate keys) to see the actual full element names and the short names they map to.

In a simple case, you would want to make sure that the elementmap for a box had 26 entries - one for each of Face1 through Face6, Edge1 through Edge12, Vertex1 through Vertex8.  Once you do operations between boxes, etc everything gets much more complex.

There are some cases where primitives do not get elementmaps until after an operation.  The best way to check that this isn't a bug is to look and see if the equivalent on LS3 has an elementmap - if it doesn't, that's because there isn't anything useful to TNP defend yet.  That might happen after the first calls here, and isn't necessarily a reason to panic.  If there's no elementmap at the end though ( and I bet there isn't ), we have no TNP mitigation awareness.

So, we want to look before and after the operations; everywhere we have a Part.Shape.  And we're just looking to see that something is still in the ElementMapSize.  Wherever it suddenly disappears - that's the broken python method which will lead us to the broken c++ implementation.

Obviously, if this is done as tests, then it will flag any future regression.  But just noting where we lose an elementmap should be sufficient to get us bug free now - thus the multiple options for how to proceed.

Thanks for the questions, keep 'em coming!

yorikvanhavre, 2024-07-02:
Indeed this:

```python
base=Draft.make_line(FreeCAD.Vector (0.0, 200.0, 0.0), FreeCAD.Vector (1000.0, 200.0, 0.0))
wall = Arch.makeWall(base, width=100.0, height=2000.0)
wall.Shape.ElementMap
```

is empty. `base.Shape.ElementMap` is also empty. I guess we have indeed to go back to the beginnings here :) I'll dig further ASAP

bgbsww, 2024-07-02:
Not surprised that either base.Shape.ElementMap  is empty; no maps on a simple line.

I'd be quite shocked if a makeCompound() call did not result in having an elementMap - early on we used this to force creation of maps so we could test their content.

So it's about the things Arch.makeWall is calling, for sure.

yorikvanhavre, 2024-07-03:
Definitely. The way forward is then identifying the different Part operations inside the construction of the wall, and check which has an ElementMap or not. I'll have a look!

yorikvanhavre, 2024-07-03:
Not sure it can be fixed for 1.0 though... @maxwxyz is it ok to remove the 1.0 label?

maxwxyz, 2024-07-03:
@yorikvanhavre I have removed the Blocker label, I'd still leave 1.0 as nice to have ;)

yorikvanhavre, 2024-07-04:
Let's try ;)

bgbsww, 2024-07-07:
I took a look at this, creating a Wall and then a Door on it, trying to start to understand.

I also looked for Shape.ElementMaps on everything, and they are mostly missing.

To the question above, element maps are all created in makeShapeWithElementMap, or mapSubElement, which are called by things like makeElementShape.

The question here seems to be what happens when you call a python makeCompound ( for example ).  The c++  makeElementCompound creates a map for the subelements.  the c++ makeCompound ( pre toponaming method ) does not.

If I'm reading the interface file correctly in Mod/Part/App/AppPartPy.cpp then the python makeCompound goes to the c++ makeCompound.  This appears to be the cast in both main and LS3.

The solution here (or part of it) is to either add makeElementX versions of these calls to the python interface and use them, or to move the python interface from using makeX to using makeElementX (which would be my recommendation, pending any other issues; let's make the interface automagically do TNP mitigation).

What needs confirmation is that the same problem exists in LS3, and there isn't some magic mitigation going on there that needs deeper research.  If it does exist there, then the research is essentially about how to make python modules able to use the element map code that isn't currently exposed at the python level.

yorikvanhavre, 2024-07-08:
What happens under the hood is 1) a face is extruded to form a wall solid, and 2) another solid is subtracted to create an opening for a door or a window. Is that something that already could be done TNP-aware? Maybe we could start there, is it something that should already be supported by TNP? I'm not sure how far

bgbsww, 2024-07-08:
> The TNP solution introduced in the current dev cycle (v0.22/v1.0) relies on a shape history. Walls based on wires and sketches do not have such a history. The intermediate shapes used to generate them are not preserved. I am currently not sure how this can be addressed.

I'm still really fuzzy here, but poking and trying to get this to click.

Looking at the underlying operations, they seem to all be TNP-aware.  The .extrudes, for example are going to TopoShapePy::extrude, and that is calling makeElementPrism.

In PD, if you create a Body and put a Sketch in it, there is nothing in the Body.Shape.ElementMap.

If you then add a Pad, the Sketch becomes the .Profile of the Pad, and the Pad is the .Tip of the body - there's a forward reference path from the Body to the Pad to the Sketch, there are faces to work with, Pad.execute calls makeShapeWithElementMap, and there's now an elementMap on the Body; further operations then act on this, and the Pad is intimately associated with the Body.  The elementmap from the Pad is the elementmap from the Body.

In BIM, if you create a Wall, either from an existing Line, or from an implicit Sketch, there is a forward path via the .Base.

If you then add a Door, it looks to be like there is only a reverse path where Door.Host is the Wall.  So Wall knows nothing about the other objects associated with it, there is no common elementmap, and they act completely independently from a TNP perspective.

Maybe that forward reference is required?  Or we need to figure out TNP in cases with reverse reference?  I'm spitballing a little here, to spur other input.

bgbsww, 2024-07-08:
I think so.  Face.extrude already does it, I think it is the independence of the Part.Face from the original Wall that is the issue.

Subtractions was empty in my original example, so that code path doesn't apply, I think.

So, if you look in TopoShapePyImp.cpp you'll see that almost everything calls makeElementX and not makeX.   Except for makeWires which calls makeWires and not makeElementWires.

Switching to makeElementWires breaks a small number of tests; not sure if it helps this issue yet.

PartDesignTests.TestTopologicalNamingProblem.testPartDesignTNPFillet
SketcherTests.TestSketcherSolver.testRemovedExternalGeometryReference
TestArch.ArchTest.testRemove

This also reminded me that the python toposhape .mapShapes()  method exists, and that it explicitly calls makeShapeWithElementMap  ( replacing the shape that python references with one with an element map ).

```
face = Part.Face(wire)
face.mapShapes()

```
might change things.

bgbsww, 2024-07-08:
The testPartDesignTNPFillet has an element map size of 62 instead of 64.  That arguably is more correct, and testPartDesignTNPChamfer which is very nearly identical has a that size of 62.

testRemovedExternalGeometryReference drops the number of edges in the threaded hole from 13 to 12.  That likely needs exploration to prove it is okay.

TestArch.ArchTest.testRemove  was an error I'd made in adding instrumentation.  It is fine.

bgbsww, 2024-07-08:
To restate the original bug ( please correct me ):

Expression references to elements do not change when the target object changes.
~BIM Element references do not change when Host object changes.~

The numbering changes are normal expected behavior, the problem is that we don't execute a lookup and change the references when that happens.

In the case of the test file, for example in Window1 the .Position.y is set to be "Wall.Shape.Vertex2.Y".  Adding to the Wall is going to change which Vertex is Vertex2 in many cases.

Is this use of expressions to set the placement of BIM subcomponents the common case, or an exceptional one?

kaiwas, 2024-07-08:
> Is this use of expressions to set the placement of BIM subcomponents the common case, or an exceptional one?

I would be glad to know that there is another way to "attach" the window to the wall that is not a straight line.
You can snap to the coordinates of the sketch vertices, but it’s not as clear as a wall. Because this will require additional mathematical calculations taking into account the width of the wall.
The example given may not be typical, but it would be very convenient if it worked.

bgbsww, 2024-07-08:
Yeah, I agree with what you are trying to do here, but I'm struggling with if this is a BIM bug, or essentially a feature request to allow TNP adjustments to occur within expressions, which might be a big, difficult, potentially slow thing to do.

The use case needs a workable solution for sure.

bgbsww, 2024-07-09:
Notes before I walk away from this one for a while:

I can't find any code in App/Expression.cpp in LS3 that leads me to believe expressions were ever made TNP mitigation aware.

importSubName functionality might be the place to add that functionality if we deemed it appropriate, but I'm nervous about both performance and behavior side effects.

yorikvanhavre, 2024-07-09:
I don't think expressions need to be TNP aware. The idea here would be that, ex. Edge6 "stays" Edge6 simply.
But it might be the responsability of BIM objects to maintain the elements table coherently. 

> So, if you look in TopoShapePyImp.cpp you'll see that almost everything calls makeElementX and not makeX. Except for makeWires which calls makeWires and not makeElementWires.

Thanks! I'll play a bit with that and report. Baby steps :)

bgbsww, 2024-07-10:
(confused still):

```
import Draft, Arch, DraftGeomUtils, Part
points = [FreeCAD.Vector(0, 0, 0), FreeCAD.Vector(1000, 0, 0)]  # we define 2 points
line = Draft.make_wire(points)  # we create a base line
wall1 = Arch.makeWall(line, height=3000, width=100)  # the wall is built

circ1 = Part.makeCircle(50)
Part.show(circ1,"circ1")
ext1 = App.ActiveDocument.addObject('Part::Extrusion','Extrude001')
ext1.Base = App.ActiveDocument.circ1
ext1.DirMode = "Normal"
ext1.DirLink = None
ext1.LengthFwd = 100.0
ext1.LengthRev = 100.0
ext1.Solid = True
ext1.Reversed = False
ext1.Symmetric = False
ext1.Placement = App.Placement(App.Vector(500.000,0.000,700.000),App.Rotation(App.Vector(1.000,0.000,0.000),90.000))
App.ActiveDocument.recompute()
print("Wall before puncture:")
print(len(wall1.Shape.Faces),"Faces")
print(len(wall1.Shape.Wires),"Wires")
print(len(wall1.Shape.Edges),"Edges")
print(len(wall1.Shape.Vertexes),"Vertexes")
print(wall1.Shape.Vertexes[0].Point, "Vertex1")
print(wall1.Shape.Vertexes[1].Point, "Vertex2")
print(wall1.Shape.Vertexes[2].Point, "Vertex3")
print(wall1.Shape.Vertexes[3].Point, "Vertex4")
print(wall1.Shape.Vertexes[7].Point, "Vertex8")
print(wall1.Shape.ElementReverseMap)
wall1.Subtractions = [ext1]
App.ActiveDocument.recompute()
print("Wall after puncture:")
print(len(wall1.Shape.Faces),"Faces")
print(len(wall1.Shape.Wires),"Wires")
print(len(wall1.Shape.Edges),"Edges")
print(len(wall1.Shape.Vertexes),"Vertexes")
print(wall1.Shape.Vertexes[0].Point, "Vertex1")
print(wall1.Shape.Vertexes[1].Point, "Vertex2")
print(wall1.Shape.Vertexes[2].Point, "Vertex3")
print(wall1.Shape.Vertexes[3].Point, "Vertex4")
print(wall1.Shape.Vertexes[7].Point, "Vertex8")
print(wall1.Shape.Vertexes[8].Point, "Vertex9")
print(wall1.Shape.Vertexes[9].Point, "Vertex10")
print(wall1.Shape.ElementReverseMap)

```
on LS3:

```
21:30:03  Wall before puncture:
21:30:03  6 Faces
21:30:03  6 Wires
21:30:03  12 Edges
21:30:03  8 Vertexes
21:30:03  Vector (0.0, 50.0, 0.0) Vertex1
21:30:03  Vector (0.0, 50.0, 3000.0) Vertex2
21:30:03  Vector (1000.0, 50.0, 0.0) Vertex3
21:30:03  Vector (1000.0, 50.0, 3000.0) Vertex4
21:30:03  Vector (1000.0, -50.0, 3000.0) Vertex8
21:30:03  {}
21:30:03  Wall after puncture:
21:30:03  7 Faces
21:30:03  9 Wires
21:30:03  15 Edges
21:30:03  10 Vertexes
21:30:03  Vector (0.0, 50.0, 0.0) Vertex1
21:30:03  Vector (0.0, 50.0, 3000.0) Vertex2
[... 46 lines cut ...]
21:36:09  <TopoShape> TopoShapeExpansion.cpp(1892): unnamed lower element Edge6
21:36:09  Arch ProcessSubShapes
21:36:09  Wall after puncture:
21:36:09  7 Faces
21:36:09  9 Wires
21:36:09  15 Edges
21:36:09  10 Vertexes
21:36:09  Vector (0.0, 50.0, 0.0) Vertex1
21:36:09  Vector (0.0, 50.0, 3000.0) Vertex2
21:36:09  Vector (1000.0, 50.0, 0.0) Vertex3
21:36:09  Vector (1000.0, 50.0, 3000.0) Vertex4
21:36:09  Vector (1000.0, -50.0, 0.0) Vertex8
21:36:09  Vector (1000.0, -50.0, 3000.0) Vertex9
21:36:09  Vector (550.0, -50.0, 700.0) Vertex10
21:36:09  {'Vertex1': 'Vertex1;:H,V', 'Vertex2': 'Vertex2;:H,V', 'Vertex3': 'Vertex3;:H,V', 'Vertex4': 'Vertex4;:H,V', 'Vertex5': 'Vertex1;:Hdee,V;:G;XTR;:Hdee:7,E;:G;CUT;:Hdef:7,V;:H,V', 'Vertex6': 'Vertex6;:H,V', 'Vertex7': 'Vertex7;:H,V', 'Vertex8': 'Vertex8;:H,V', 'Vertex9': 'Vertex9;:H,V', 'Vertex10': 'Vertex1;:Hdee,V;:G;XTR;:Hdee:7,E;:G2;CUT;:Hdef:8,V;:H,V', 'Face1': 'Face1;:H,F', 'Face2': 'Face2;:H,F', 'Face3': 'Face3;:H,F', 'Face4': 'Face4;:H,F', 'Face5': 'Face5;:H,F', 'Face6': 'Edge1;:Hdee,E;:G;XTR;:Hdee:7,F;:M;CUT;:Hdef:7,F;:H,F', 'Face7': 'Face7;:H,F', 'Edge1': 'Edge1;:H,E', 'Edge2': 'Edge2;:H,E', 'Edge3': 'Edge3;:H,E', 'Edge4': 'Edge4;:H,E', 'Edge5': 'Edge1;:Hdee,E;:G;XTR;:Hdee:7,F;:G;CUT;:Hdef:7,E;:H,E', 'Edge6': 'Edge6;:H,E', 'Edge7': 'Edge7;:H,E', 'Edge8': 'Edge8;:H,E', 'Edge9': 'Edge9;:H,E', 'Edge10': 'Edge10;:H,E', 'Edge11': 'Edge11;:H,E', 'Edge12': 'Edge12;:H,E', 'Edge13': 'Edge13;:H,E', 'Edge14': 'Vertex1;:Hdee,V;:G;XTR;:Hdee:7,E;:M;CUT;:Hdef:7,E;:H,E', 'Edge15': 'Edge1;:Hdee,E;:G;XTR;:Hdee:7,F;:G2;CUT;:Hdef:8,E;:H,E'}

```

So, the Vertexes get renumbered.  That's inevitable.  The toponaming mitigation looks reasonable at this stage, with an elementMap created as soon as we do a subtraction operation on the wall.

If we want the actual example from this bug report to work, we have to do something about TNP in expressions, because that's what it is using.

If we're looking for some alternative path for a good BIM workflow that allows adding openings to a wall that are parametrically associated with that wall, then we need to define what the approach is, and then can look to see how to fit it into the framework here.  But that would not be this bug.

As I started with, I'm confused.  This bug looks dead-on-arrival to me; everything is functioning as it is supposed to.  It seems to me that if you want to refer to the locations in the wall in expressions, you need to do that without using the geometry within the wall (vertexes, etc), and by using overall geometry like the placement of the wall.   What silly thing am I missing?  Maybe there's a common workflow that I still don't get?

yorikvanhavre, 2024-07-11:
This is good. Now what do you think of this @bgbsww :

1. Do you think that if we had an ElementMap before the punction, we would get a better result?
2. Consequently, and since we have good control of how the wall is built, would it make sense - or is it possible - to build the ElementMap ourselves? I mean, we have the base face, so we know which one it is, so we could quite easily define an order ourselves

yorikvanhavre, 2024-07-11:
I rescheduled this for post 1.0 because the scope is obviously bigger than a simple bugfix

bgbsww, 2024-07-11:
Thanks, @yorikvanhavre 

1. No, I don't think we'll get a better result, because an ElementMap is not going to solve referring to changing elements using fixed textual names.  We would have to use links for the ElementMap to be relevant, or write new code to look up and register our uses of elements.
2. Yes, we could build the ElementMap, but see questions below; I'm not sure how it would be used.

I guess I still don't understand if / where / how the positional connections are made between a wall and it's subtractives.  ~That's what we need to focus on, since we need to ensure that uses mapped names as appropriate in order for the ElementMap to become useful.~

If the subtractives are just independent DocumentObjects that happen to intersect the wall and be listed in the Subtractives list, I don't see how this would work - there is nothing for the elementMaps to operate on.

Actually, I guess that's making an assumption.  Let me try it this way:  I assert that there is no reasonable way to prevent the numbering of geometry in a wall from changing and there never will be, so eliminate that possibility.  **Now, what is the need that has us caring about if those change?**

So far, that seems to be referring to them in expressions, and that's a bad idea, since we are referring to something that is likely to change out from under us.  The strength of expressions is that they let you do rash things that weren't anticipated. but YMMV.

**Is there another place the element numbering / names are used in BIM?** (A cursory search doesn't show it, but I really don't know either the code or the user experience more than the poking I've done the last few days)

**It does seem we need a mechanism that exposes a stable reference that can be used in expressions - is the existing property sufficient?**  By which I mean use  Wall.Base.Shape.Vertex1 (perhaps with some offset values) instead of Wall.Shape.Vertex1 since the underlying geometry the wall is based on should not change in response to subtractives added to the wall.  If you are changing that underlying wire, you should expect to have to do work to adjust objects that reference it, although it is much, much less likely that the Vertex names will change in a way that troubles you due to the nature of 2D geometry.

yorikvanhavre, 2024-07-16:
> If the subtractives are just independent DocumentObjects that happen to intersect the wall and be listed in the Subtractives list, I don't see how this would work - there is nothing for the elementMaps to operate on.

Indeed you are describing exactly what happens here. In BIM it's like in Part, there is no positioning relationship. You model objects, position them manually, then intersect them when needed.

But this is also what happens in Part Cut. Does that mean only PartDesign will have TNP support? I supose no, but I  understand PartDesign is where you've been focusing up to now.

I think - and that's also the most reliable path, I guess - we'll need to build the elements map ourselves in BIM. Identify the base face of the wall, identify the extruded faces, identify the cuts... All this is doable and not that complicated I  guess.

Would you help me to build a simple case? Say we have a base face that gets extruded along a vector:

```python
v1 = FreeCAD.Vector(0, 0, 0)
v2 = FreeCAD.Vector(10, 0, 0)
v3 = FreeCAD.Vector(10, 10, 0)
v4 = FreeCAD.Vector(0, 10, 0)
pol = Part.makePolygon([v1, v2, v3, v4, v1])
baseface = Part.Face(pol)
# so far so good, one face, 4 verts and 4 edges.
# How could we already create an element map here, and populate it?
extrusion = baseface.extrude(FreeCAD.Vector(0, 0, 10))
# Here we would calculate everything we need, the correct order of verts, edges and faces
# Same question, we have our right order, how do we populate the elements map?
```

I guess I basically need some docs on how to build/update an element map in Python... :sweat_smile:

bgbsww, 2024-07-16:
I may wander here a little: dumping information as I get to trying to answer your question:

TNP support should be in all of Part and PartDesign, and is in Sketcher with horrendous bugs I'm chasing down.  Part Cut relies on Part Boolean, and here's a snippet of that c++ code:

```
#ifndef FC_USE_TNP_FIX
        std::vector<ShapeHistory> history;
        history.push_back(buildHistory(*mkBool, TopAbs_FACE, resShape, BaseShape));
        history.push_back(buildHistory(*mkBool, TopAbs_FACE, resShape, ToolShape));

        if (this->Refine.getValue()) {
            try {
                TopoDS_Shape oldShape = resShape;
                BRepBuilderAPI_RefineModel mkRefine(oldShape);
                resShape = mkRefine.Shape();
                ShapeHistory hist = buildHistory(mkRefine, TopAbs_FACE, resShape, oldShape);
                history[0] = joinHistory(history[0], hist);
                history[1] = joinHistory(history[1], hist);
            }
            catch (Standard_Failure&) {
                // do nothing
            }
        }

        this->Shape.setValue(resShape);
        this->History.setValues(history);
        return App::DocumentObject::StdReturn;
#else
        TopoShape res(0);
        res.makeElementShape(*mkBool, shapes, opCode());
        if (this->Refine.getValue()) {
            res = res.makeElementRefine();
        }
        this->Shape.setValue(res);
        return Part::Feature::execute();
#endif
```
The key here is in the replacement code we are making a TopoShape ( Not a TopoDS_Shape ), and then calling makeElementShape which properly creates an elementmap.

Go into Part, click the Box and the Cylinder, select them, make a Cut from them and App.ActiveDocument.Cut.Shape.ElementMap will reward you with an element map.  .Box and .Cylinder won't have any, since they are primitives.

I don't know of any prior art for creating an element map from python, which means this is going to be subject to lots of potential bugs, both in the original LS3, and then with additional possibilities in the porting.  The .mapSubElements ( not .mapShapes, that's something else ) should be the entry point here.

But that is only going to work if we have sub elements, and we don't in BIM.

The only way I know to create element maps is with a call to makeShapeWithElementMap as called by makeElementShape as called by all of the Part operations.  Those Part operations are then called by PartDesign operations.  

I don't see any python methods to call these lower level methods, so I think the thing you are asking for does not exist.  However, even if we had an element map, I can't see how it would really help here, since we have to look into the element map at every usage, and in this case, the usage is within an expression.

I guess I'm just continuing to miss something here - what is the overall concept here of what TNP aware means in BIM?  Right now in this specific bug report, I see a a misuse of expressions to reference something volatile with the expectation that it won't change and offered a different approach above.  **As long as that is the case, and we don't add lookups in expressions, then nothing else we do will change this, no?**

TNP aware does not mean that element names will not change.  It means that we adjust correctly when they do change.  Expressions do not currently support this.

Tracing that python .extrude() call above in c++ debugger goes through TopoShape::makeElementPrism as one would expect, that calls makeElementShape path and  it determines that the shape being extruded ( the Part.Face ) is not mappable and exits without a map.  That would be alarming if LS3 didn't do exactly the same thing.  Using makeFace instead has the same effect.  Since we know that in c++, extrusions of faces result in things with elementMaps, that points at something wrong.

The reason that the shape being extruded isn't mappable is because it has a zero Tag.  Zero tags are okay in the target object (which is the return), but any source object without one will be ignored in mapping.  **Changing the code above to set `baseface.Tag = 1` before the extrusion results in an object with an ElementMap.**

kaiwas, 2024-07-18:
There was hope that if you “linked” the windows to the wall sketch, the model would not break.
But this still happens if you add another line to an existing sketch (or split an existing line into segments).

In general, this approach is more stable if you do not change the sketch. But it’s more complicated because, in addition to the coordinates of the sketch vertices, you have to write the wall width and offset into expressions.

https://github.com/user-attachments/assets/fb344743-c1fa-4f80-8b3d-4816dfd64c57

[Test_Wall_TNP_2.FCStd.del_this_and_zip.zip](https://github.com/user-attachments/files/16278871/Test_Wall_TNP_2.FCStd.del_this_and_zip.zip)

paullee0, 2024-07-20:
Just noted interesting discussion here :)

Haven't read through the whole thread yet, but this is one feature the SketchArch Add-on try to provide and there is explanation how the way it is implemented ('toponaming-tolerant' but not relay on FC 1.0 Core Toponaming feature), in gist:

- SketchAddon allow user to attach Arch Objects (e.g. Window/Door, Equipment) to ArchWall 'parametrically' and 'toponaming-tolerant'
- It use Sketcher::SketchObjectPython (rather than ordinary Sketcher::SketchObject), it records the Tag (semi-persistant) of the Sketch Edge the ArchObjects attach to
- Thus it is 'toponaming-tolerant', edited the ArchWall / the underlying ArchSketch, it won't break the attachment

See links below if interested:-
- https://github.com/paullee0/FreeCAD_SketchArch?tab=readme-ov-file#2-parametric-placementattachment-of-arch-objects-intuitively
- https://github.com/paullee0/FreeCAD_SketchArch/blob/master/README.md#7-topological-naming-tolerant
- https://forum.freecad.org/viewtopic.php?p=600105#p600105

Hope it helps :)

bgbsww, 2024-07-20:
It does, thanks.  I think we might need to rename this or close it - not because we don't need an obvious solution, but because changing numbering of element is definitely not a bug, and referring to them in an expression expecting them not to change is not an appropriate approach.  It appears that there is a working approach, and maybe this wants to be integrated into BIM.  This doesn't seem to have anything to do with an error in the underlying TNP code.  I'm going to unassign until something TNP related crops up..

paullee0, 2024-07-20:
Some video demos :)

[YT Video Demo 1](https://www.youtube.com/watch?v=Sqt2rezgPZk&t=9s)

[YT Video Demo 2](https://www.youtube.com/watch?v=_d-BZJH70xk)

[YT Video Demo 3](https://www.youtube.com/watch?v=sp5tACZWkV4)

paullee0, 2024-07-20:
> There was hope that if you “attached” the windows to the wall sketch, the model would not break. But this still happens if you add another line to an existing sketch (or split an existing line into segments).
> 
> In general, this approach is more stable if you do not change the sketch. But it’s more complicated because, in addition to the coordinates of the sketch vertices, you have to write the wall width and offset into expressions.

With ArchSketch, it won't break either user add edges or delete edges in the ArchSketch :)

paullee0, 2024-07-21:
> Tracing that python .extrude() call above in c++ debugger goes through TopoShape::makeElementPrism as one would expect, that calls makeElementShape path and it determines that the shape being [...]

I read again and again and do not seem to understand all the technicality of the TNP solution behind :)

Maybe anyone has the spare time to test it LS3 has some solution e.g. on a simple straight ArchWall?

bgbsww, 2024-07-21:
I should have gone back and edited that.  The reason that the Part.Face was not mappable was due to the missing Tag.  The behavior is the same in LS3, because the python code for BIM does not set the Tag.

kaiwas, 2024-07-21:
Did I understand correctly that there is no solution at the moment? And after the change, the model will still remain a “Schrodinger’s cat”? (Maybe it will break, maybe not)

paullee0, 2024-07-25:
It seems nobody has a concrete answer at the moment?  :)

 @yorikvanhavre,  see if you have a gap to check 3 approaches to the problem in https://github.com/paullee0/FreeCAD_SketchArch/blob/master/README.md#7-topological-naming-tolerant
(the 3rd approach just mean Realthunder TNP LS3)

1st approach was implemented and briefly described above, quoted below.  

2nd approach was about Sketch Geometry Extension - blocked by a 'bug' (https://github.com/FreeCAD/FreeCAD/issues/15545)

> Just noted interesting discussion here :)
> 
> Haven't read through the whole thread yet, but this is one feature the SketchArch Add-on try to provide and there is explanation the way it is implemented ('toponaming-tolerant' but not rely on FC 1.0 Core Toponaming feature), in gist:
> 
>     * SketchAddon allow user to attach Arch Objects (e.g. Window/Door, Equipment) to ArchWall 'parametrically' and 'toponaming-tolerant'
> 
>     * It use Sketcher::SketchObjectPython (rather than ordinary Sketcher::SketchObject), it records the Tag (semi-persistant) of the Sketch Edge the ArchObjects attach to
> 
>     * Thus it is 'toponaming-tolerant', edited the ArchWall / the underlying ArchSketch, it won't break the attachment
> 
> 
> See links below if interested:-
> 
>     * https://github.com/paullee0/FreeCAD_SketchArch?tab=readme-ov-file#2-parametric-placementattachment-of-arch-objects-intuitively
> 
>     * https://github.com/paullee0/FreeCAD_SketchArch/blob/master/README.md#7-topological-naming-tolerant
> 
>     * https://forum.freecad.org/viewtopic.php?p=600105#p600105
> 
> 
> Hope it helps :)

yorikvanhavre, 2024-07-30:
I think what we aim at here, at BIM level, is to have reliable subobject names across the different "stages" of development of an arch object. Typical example: We have a simple wall which is a simple rectangular shape with 6 faces. Then we add windows, we move doors, etc.. many times over the course of the project. And we would like the "border" edges and the "inner" edges of the wall to remain with the same name, no matter what you union or subtract or un-union or un-subtract.

Said like this, it seems to me to be a typical TNP case.

Of course, we have specific ways of working in BIM, which are not the typical Sketch->PartDesign workflow, and I don't think this falls into @bgbsww's work scope. I see this rather as a first discussion on how to approach the TNP thing BIM-wise.

The thing is, typical BIM modeling uses a lot of smaller objects, and has rarely very complex parametric chains. The most advanced case is probably the one we're discussing here, the wall with holes in it. And we have very high possible control over which edge should be what. So I think there are here two goals:

1. Allow python modules like BIM to ~~mess with~~ handle the element map
2. Build logic in BIM to handle the element map

As @bgbsww said there is currently no python functionality in the FreeCAD code, so there is no way to attack this right now. And TBH I'd wait a bit for TNP stuff to stabilize and we get a clear idea of what works and what doesn't specifically regarding Part booleans.

In any case, I propose we leave this open for reference, but it probably won't get fixed soon.

> @yorikvanhavre, see if you have a gap to check the potential with 2 available approaches to the problem in https://github.com/paullee0/FreeCAD_SketchArch/blob/master/README.md#7-topological-naming-tolerant

I didn't understand everything but it seems to me we need to understand better what currently works. In any case, we have a TNP remediation structure in place now, that's definitely what we need to use...

bgbsww, 2024-07-30:
Thanks, Yorik.

> holes in it. And we have very high possible control over which edge should be what. So I think > there are here two goals:

I assume this would be by effectively redrawing shapes with the edge order what you want it to be whenever you do a BIM operation?  That would sidestep the TNP since there would ideally never be changes - only additions; if you changed or deleted something like a window ( or a simple hole ) after the fact, and something referred to it, then that would break, but it has to, TNP or no.  The tricky part is if you ended up referring to the wrong thing.  If you have two openings in a wall, and are referring to the first one, and then delete it, the references are going to "jump" to the second one as it inevitably gets the edges renumbered down, since we don't support gaps in the numbering (and likely could not ever, that's from occt on up).  So you would likely need to detect those cases and do something.

> 1. Allow python modules like BIM to ~mess with~ handle the element map
> 2. Build logic in BIM to handle the element map
>
> As @bgbsww said there is currently no python functionality in the FreeCAD [...]

I think that python Toposhape's .mapSubElements should give the window in here - that python functionaility exists, but I see no evidence that it has ever been used thus I would initially be skeptical about any results until proven.  If that doesn't work for the BIM model due to not having that subElement relationship [it's reversed, right?], then we are in territory for which there is not only no python interface, there is **also no c++ code**.   In early test writing days, it was very frustrating to not be able to create an elementMap for an element, only for subelements.  

So when this starts to be addressed some clarification on what to expose and discussion on how to do it should commence.  I'm up for at least that discussion when we get there.

paullee0, 2024-07-30:
> I didn't understand everything but it seems to me we need to understand better what currently works. In any case, we have a TNP remediation structure in place now, that's definitely what we need to [...]

Yes, difficult to explain indeed.

Anyway in short, with the SketchArch add-on current implementation, attaching Windows/Doors to a Wall won't suffer from 'TNP' problem.

Hope there are more tests to prove this works :)

kaiwas, 2024-07-30:
> The most advanced case is probably the one we're discussing here, the wall with holes in it.

I completely agree with you that there are not too many "modifications" for walls. But I want to add that in addition to windows and doors, this can be an interfloor ceiling resting on the wall.

[image]

paullee0, 2024-07-30:
> In general, you can attach windows and doors to the sketch on which the wall is built. As long as we do not change the sketch much, we have a stable object that does not depend on holes. But I can say that this approach is difficult because you have to independently calculate thicknesses, offsets and other parameters inherent in the wall. But if there was a way to simplify such calculations, it would be much easier and you could not worry about the presence of holes in the wall.
>

This is the base concept in the SketchArch add-on, implemented :)  And each edge is traced, even deleting or adding an edge would not result in attachment jumping.  Offset, thickness etc are handled.

kaiwas, 2024-07-30:
> This is the base concept in the SketchArch add-on, implemented :) And each edge is traced, even deleting or adding an edge would not result in attachment jumping. Offset, thickness etc are handled.

Good!
Someday I will definitely get to serious testing of this add-on )))
It will probably be in winter. In the warm season I have a lot of work and absolutely no time.

yorikvanhavre, 2024-07-31:
> I completely agree with you that there are not too many "modifications" for walls. But I want to add that in addition to windows and doors, this can be an interfloor ceiling resting on the wall.

Of course there are further developments possible, but in general terms, the requirements for complex parametric chains are currently fewer in BIM than in other disciplines. That does not mean it would not be handy indeed. And, with more reliable toponaming, maybe we'll get more parametric-hungry too :)

> I assume this would be by effectively redrawing shapes with the edge order what you want it to be whenever you do a BIM operation? That would sidestep the TNP since there would ideally never be [...]

I think you got it totally right. Basically this is the main scenario I have in mind:

1. You draw a rectangle. Edges 1, 2, 3 and 4 are obviously the 4 rectangle's edges
2. You extrude that rectangle into a wall. Edges 1, 2, 3 and 4 are still the same edges as in the rectangle, in other words the wall's bottom face. Edges 5 to 11 are the other edges. For example, 5 to 8 are the vertical edges, 9 to 12 are the top face borders. The rectangle might change often as you often push/pull the wall extremities to "meet" other walls, or want to change its thickness
3. You subtract a volume to create an opening in the wall (a window). Edges 1-4 are still the bottom face, 5-8 still the vertical edges, 9-12 still the top face edges, and 13-24 are the edges around the window hole
4. What architects usually want is the window opening to be placed at, say, 100cm from the left border of the wall. So you would make the subtracted volume as an extruded sketch, and place that base sketch on the wall's face, and reference, say, Edge5 for the wall and place a 100cm constraint there. So far so good.
5. In most cases we'll want to reference one of the four main vertical edges of our wall (let's say Edges 5 to 8). These should stay as much as possible throughout the life of the wall. They are basically the main edges of our wall. So we can hook things to them: openings, but also other elements that need to lie on the wall, such as beams and slabs

That's basically the "canonical case", I would say. However it's indeed not that simple:

1. Suppose now, one might want to add a second window, placed at 50cm from the first one. This is interesting as if the first one moves, the second will stay at 50cm no matter what. What if the first one is removed then? Indeed the model should break, TNP or not.
2. What if one adds a door? It will "break" our bottom face into two. That's where some "TNP magic" would be handy: Edges 1 and 3 would change, but Edges 5 to 8 should stay the same. Additional edges from the new bottom face should be added at the end of the list

There is here I guess, some part that "should be magically handled" by TNP, and others that should be hand-set by BIM. It is  indeed not clear to me at this point which is one and which is the other.

Anyway, I think thins issue is getting a really interesting reference discussion on BIM and TNP. Maybe it's worth renaming or turning into a github discussion?

kaiwas, 2024-07-31:
> renaming or turning into a github discussion?

Oh no! Just don't turn it into a discussion. (Nobody sees them)
I have the impression that "discussion" on GitHub is some kind of graveyard of ideas.
This is clearly evident from the number of messages in the "discussion" thread.

yorikvanhavre, 2024-08-01:
Okay ;) Let's keep this one here then

paullee0, 2024-11-22:
A video posted on Youtube earlier back in 2021 showcasting how TNP-tolerant works in the add-on for Arch/BIM.
(https://www.youtube.com/watch?v=7amrcUmd1YA](https://www.youtube.com/watch?v=Sqt2rezgPZk))

https://github.com/user-attachments/assets/c4721f4d-114f-41a3-b5ed-3fc436b638b9

maxwxyz, 2025-03-03:
@yorikvanhavre might the new wall / window tools solve this? At least the 6 base faces of a straight wall segment should be known?

yorikvanhavre, 2025-03-04:
It won't, I'm not touching TNP.

That's something we should attack at some point, because indeed in 90% of the walls we create the geometry ourselves and can control verts/edges order.  Just not sure how to interact with the ElementMap form Python yet, I think nobody has taken that path much yet...

kaiwas, 2025-03-04:
> can control verts/edges order.

If some new algorithms can make it so that windows and doors do not fall out of the walls when the walls are moved, then the problem of toponyming will not be so important.
But at the moment, the only way to "fix" windows in the wall is expressions. And expressions can very easily break if they use data from the "parent" geometry.
A partial solution to the problem is to bind to mathematical data (for example, VarSet). But this is much more labor-intensive than simply inserting a window into a wall.

paullee0, 2025-03-04:
Looks intuitive and handy indeed.  The basic functionalities work similarly with Sketch as Base, and with ArchSketch, Window/Door objects can attach parametrically and TNP-tolerant; all the GUI features are those make it very user-friendly, hope there are peoples who would like to add these to FC?

kaiwas, 2025-03-05:
> ArchSketch

Can you add to your component so that the window is inserted to the specified height (window sill), and not falls to zero?
And also - put all SketchArch settings in a separate tab? With them, the property panel has to be scrolled a lot.

paullee0, 2025-04-18:
Check this PR / Forum Discussion:

- https://github.com/FreeCAD/FreeCAD/pull/20872
- https://forum.freecad.org/viewtopic.php?p=822493#p822493

paullee0, 2025-05-11:
> Check this PR / Forum Discussion:
> 
>     * [[BimWindow] Integrate Interactive Creation Tool w/ SketchArch Parametric Attachment Feature #20872](https://github.com/FreeCAD/FreeCAD/pull/20872)
> 
>     * https://forum.freecad.org/viewtopic.php?p=822493#p822493
> 
> 
> Parametric.Placement.Feature.10.-.Stock.BIM.UX.Integration.mp4

This PR was merged, IMHO, the 'problem' in this issue is addressed.

And with this PR, https://github.com/FreeCAD/FreeCAD/pull/21261, all the existing behaviour is preserved whilst there is no 'TNP' problem.

kaiwas, 2025-05-11:
> This PR was merged, IMHO, the 'problem' in this issue is addressed, no more TNP problem headache.
> 
> And with this PR, [#21261](https://github.com/FreeCAD/FreeCAD/pull/21261), all the existing behaviour is preserved whilst there is no 'TNP' problem.

A big statement ) We need to do some testing.
What happens if we make the base-sketch on the left into the sketch on the right?

[image]

But overall I agree that SketchArch expands the possibilities and stabilizes the situation.

[8 more replies]