FreeCAD/FreeCAD #14928: [TNP] [BIM] Assigning a multimaterial to walls changes the numbering of vertices/edges/faces

kaiwas, 2024-06-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In this example, the window coordinates use expressions that take values ​​from the coordinates of the wall vertices.
When we assign a multimaterial to a wall, making it multilayer, the numbering of all subcomponents changes..

To reproduce the problem, simply assign the multimaterial to the wall. An example is shown in the video.

https://github.com/FreeCAD/FreeCAD/assets/32273833/e0d6a667-6394-4053-bf92-35c75ea4be38

Test file is attached. Change the extension .txt to .FCStd
[Test_Walls_and_Windows.txt](https://github.com/user-attachments/files/15966119/Test_Walls_and_Windows.txt)

### Full version info

```shell
0,22dev
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

bgbsww, 2024-07-01:
This is likely code changes to BIM, which is outside of what I know.  Is there a BIM expert who can collaborate on this?

FEA-eng, 2024-07-09:
> Is there a BIM expert who can collaborate on this?

@yorikvanhavre ?

bgbsww, 2024-07-09:
Thanks, he is colllaborating on the closely related #14927.  I question the premise here that numbering of elements should not change, given that is fundamental to TNP mitigation.

This is about using expressions that contain explicit references to specific vertices.  Other places that refer to elements do so via links, and the TNP mitigation code automagically adjusts the target of those links when needed.  That does not happen inside of manually entered expressions, and it isn't clear yet what the right solution is.

Given that this is the same root cause as #14927, should we possibly be closing this?

yorikvanhavre, 2024-07-09:
Yes, it's basically a duplicate.

yorikvanhavre, 2024-07-09:
Duplicate of #14927

closed 2024-07-09 by yorikvanhavre