FreeCAD/FreeCAD #14929: GUI: Custom themes have bad discoverability

kadet1090, 2024-06-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Prior to 0.22/1.0 FreeCAD shipped with few pre installed themes giving people multiple styles to choose. While generally it is great to provide user with choice it is problematic to maintain so much and so most of them were quite quickly abondoned with only occasionall fixes. That is why for 0.22/1.0 it was decided to include in core only bare minimum number of themes: light, dark and classic one. The rest will be moved to addon repository.

Because of that we should increase visibility of custom themes in preferences. My suggestion is to simply add button "find more" close to theme selector so everyone will be informed that there are multiple other choices that are simply not pre installed anymore.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

relevant issues PRs:
- https://github.com/FreeCAD/FreeCAD/pull/13772
- https://github.com/FreeCAD/FreeCAD/pull/14827

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2024-06-25:
Good initiative!
Would like this button above the theme selection in the startpage as-well!

microbearsmb, 2024-06-26:
Agreed! If possible if this could be implemented before 1.0, it would alleviate a lot of the pain for users of these custom themes when they find them no longer pre-installed.

kadet1090, 2024-06-26:
I hope that I'll be able to do that today, the goal is to ship it with 1.0

commit 45dc70c5 (2024-08-05): Gui: Add information about custom themes to settings

Fixes: #14929