FreeCAD/FreeCAD #14930: Default Link_Copy_On_Change setting (for links)

kaiwas, 2024-06-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Links in freecad have a powerful feature that allows you to change the appearance of the link object.
But in order to be able to change a link object, you have to look for the Link_Copy_On_Change setting in the properties window every time and change its value.
If we need to make 20 reference objects, we will have to perform this operation 20 times.
Video example

https://github.com/FreeCAD/FreeCAD/assets/32273833/ed008757-e950-459c-868d-5ebdbebe6cef

I suggest making a setting for link objects where you can specify the desired parameter.
I don't know where this setting would be most logical. Global settings are possible in the program (picture 1).

[image]

Or maybe at the level of the object from which we create links (picture 2)..
[image]

### Full version info

```shell
any versions
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

krushia, 2024-06-28:
There is a way to do this already, but I can't remember how. The code is around here:

https://github.com/FreeCAD/FreeCAD/blob/3b8f615f7a90805682fed6a306510a041a69cca8/src/App/Link.cpp#L117-L121

kaiwas, 2025-04-11:
The problem is solved with a macro. But I think it would be useful at the core level.

```
def set_link_copy_on_change():
    doc = FreeCAD.ActiveDocument
    if not doc:
        FreeCAD.Console.PrintWarning("No active document found!\n")
        return
    
    count = 0
    for obj in doc.Objects:
        if obj.isDerivedFrom("App::Link") or hasattr(obj, "LinkedObject"):
            try:
                if hasattr(obj, "LinkCopyOnChange"):
                    obj.LinkCopyOnChange = True
                    count += 1
                    FreeCAD.Console.PrintMessage(f"Set LinkCopyOnChange for: {obj.Name}\n")
                else:
                    FreeCAD.Console.PrintWarning(f"Object {obj.Name} has no LinkCopyOnChange property\n")
            except Exception as e:
                FreeCAD.Console.PrintWarning(f"Failed to set property for {obj.Name}: {str(e)}\n")
    
    FreeCAD.Console.PrintMessage(f"Done. Updated {count} objects.\n")

set_link_copy_on_change()
```