FreeCAD/FreeCAD #14938: Material: DiffuseColor property not handled properly

Roy-043, 2024-06-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The old DiffuseColor property is not handled properly in the current dev version. The transparency is ignored. Basically the current conversion results in:
`[m.DiffuseColor for m in vobj.ShapeAppearance] == vobj.DiffuseColor => True`

It should however be:
`[m.DiffuseColor[:3] + (m.Transparency, ) for m in vobj.ShapeAppearance] == vobj.DiffuseColor => True`

The attached [file](https://github.com/user-attachments/files/15969922/window-color-test.FCStd.zip) contains an Arch Window created in v0.21.
1. Remove the zip extension.
2. Open the file in V0.22.
3. Recompute (not required I think).
4. Result: the solid representing the glass has lost its transparency.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37841 (Git)
Build type: Release
Branch: main
Hash: 5f761c758e2a0164b00a7183b9c0cbe32688e123
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-01:
@Roy-043 is this still an issue? Should be fixed in tomorrows weekly.

Roy-043, 2024-07-02:
This bug still exists in the latest weekly.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git)
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

chennes, 2024-08-31:
Is this a BIM-specific bug, or does it occur throughout?

Roy-043, 2024-08-31:
It occurs throughout.

Note that setting the transparency of individual faces via the GUI was added as a feature in V0.20. Removing that in V0.22/V1.0 would be a bit strange.

Please give me some advance warning if it is not possible to solve this before the V1.0 release so that I can prepare some PRs with workarounds (at the very least for the transparency of glass in windows and doors).

chennes, 2024-08-31:
I am going to look at it tonight. Do you have time to write a short macro that demonstrates the failure that I can turn into a regression test?

davesrocketshop, 2024-08-31:
Lurking...

Roy-043, 2024-09-01:
Suggestion for test file:
```python
# import Test, materialtests.TestMaterialDocument
# Test.runTestsFromClass(materialtests.TestMaterialDocument.DocumentTestCases)

import unittest

import FreeCAD

class DocumentTestCases(unittest.TestCase):
    """
    Test class for FreeCAD material tests that need a document
    """

    def setUp(self):
        self.doc = FreeCAD.newDocument()

    def tearDown(self):
        FreeCAD.closeDocument(self.doc.Name)

    def testApplyDiffuseColor(self):
        """ Test that applying a DiffuseColor with transparency results in a correct ShapeAppearance """
        dif_col_1 = (1.0, 1.0, 0.0, 0.0)  # yellow 0% transparent
        dif_col_2 = (1.0, 0.0, 0.0, 0.5)  # red 50% transparent
        dif_col = [dif_col_1] + [dif_col_2] + 4 * [dif_col_1]

        obj = self.doc.addObject("Part::Box")
        vobj = obj.ViewObject
        vobj.DiffuseColor = dif_col

        # Workaround code:

        # sapp = []
        # for col, mat in zip(dif_col, vobj.ShapeAppearance):
            # mat.Transparency = col[3]
            # sapp.append(mat)
        # vobj.ShapeAppearance = sapp

        self.assertEqual(
            [m.DiffuseColor[:3] + (m.Transparency, ) for m in vobj.ShapeAppearance],
            dif_col
        )
```

davesrocketshop, 2024-09-01:
Just to clarify, what you're doing is setting the alpha channel on the diffuse color, not the transparency for the object. They're not the same thing. Remember that ShapeAppearance has multiple colors (emissive, specular, ambient, and diffuse). Alpha channel changes don't and shouldn't have any effect on the transparency

chennes, 2024-09-01:
@davesrocketshop so to close this issue we need two things, if I am understanding correctly. We need a UI for setting the transparency of a material (that is, we need a new widget added to the Material Settings dialog), and we need the Glass material settings that seem to be what BIM is using for window transparency to use the ShapeAppearance.Transparency property. Is that correct?

chennes, 2024-09-01:
@Roy-043 I don't think your suggested workaround is actually a "workaround" at all -- it's how it needs to be done until BIM can get updated to retain transparency as a separate property instead of treating it as part of DiffuseColor. So if you've got an idea for implementing that, I'd say go ahead with your PRs. I poked at it a bit today but I have basically zero knowledge of Arch/BIM, so I couldn't make it work.

Roy-043, 2024-09-01:
The alpha channel of the diffuse color is used to define the transparency of faces in V0.21 and previous versions. To correctly handle objects with face transparency, and also to support code that relies on this (aka backwards compatibility) the alpha channel of the DiffuseColor must be translated to the transparency of ShapeAppearance materials. Obviously the DiffuseColor and the ShapeAppearance must then have an entry for each face of the object.

Irrespective of what we do for the BIM WB, the DiffuseColor should be handled properly. Currently it is not.

chennes, 2024-09-01:
I must be missing some key principle -- we have a specific property for transparency, and we have three different colors, *all* of which can have an alpha channel. Presumably there is some historical reason for the decision to use the alpha channel of diffuse color for per-face transparency, but it's not clear to me that going forward it's reasonable to just override transparency with diffuse alpha. Do we need a migration script or something?

Roy-043, 2024-09-04:
In V0.21 and earlier we have `ShapeColor`, `Transparency` and `DiffuseColor`. The first two properties control the appearance of the whole object. `DiffuseColor` has to be used for color-per-face and also transparency-per-face.

In V0.22 the new `ShapeAppearance` and the old `DiffuseColor` are kept in sync for the RGB values. If you change either property the other changes as well. This mechanism has to be extended to include transparency values.

Roy-043, 2024-09-05:
I already have a solution for the transparency of windows prepared. I have had that ready for some time now. I have left off making a PR hoping this issue would be fixed, which would be better IMO.

I'll create a PR this week.

Roy-043, 2024-09-17:
There is an issue with backward compatibility that needs to be addressed if the alpha values of the DiffuseColor are to be ignored in V1.0.

Open this [file](https://github.com/user-attachments/files/17025485/box-1-face-transparent.FCStd.zip) in V0.21 and V1.0 (remove the zip extension). In V1.0 the top face of the box is not transparent.

Another issue is that because the mentioned alpha values are ignored, the Draft `get_diffuse_color` function needs to be revised and the code that relies on it needs to be checked.

Create a Draft_OrthoArray, with the "Link array" checkbox unchecked, from the part in [this](https://github.com/user-attachments/files/17025740/part-with-transparent-object.FCStd.zip) file in V0.21 and V1.0. In V1.0 the cylinder is not transparent in the array.

When I tried revising this function, I got stuck because I encountered issues with the color handling of Links. I'll create a separate issue for that.

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38728 (Git)
Build type: Release
Branch: main
Hash: 9de78e27f4148e8032594788a9804486e769388b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

Roy-043, 2024-09-17:
There is also a related issue with STEP import and export.

Forum topic:
https://forum.freecad.org/viewtopic.php?t=90684
KiCad issue:
https://gitlab.com/kicad/code/kicad/-/issues/18729

Open this [file](https://github.com/user-attachments/files/17035006/cube-with-1-colored-face.FCStd.zip) in V1.0 (remove the zip extension).
Select the box and export to STEP.
Check the file with a text editor: 5 faces have `SURFACE_STYLE_TRANSPARENT(1.)`.

Open the created STEP in V0.21 to verify the transparency of the 5 faces. Open the STEP in V1.0 and notice that the transparency is ignored.

chennes, 2024-09-20:
> There is an issue with backward compatibility that needs to be addressed if the alpha values of the DiffuseColor are to be ignored in V1.0.

Thanks for opening those Issues: does this one (#14938) now reduce to the above backwards-compatibility discussion?

Roy-043, 2024-09-25:
Is #16692 included in the version listed below? I am asking because the issue seems not to be solved.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38830 (Git)
Build type: Release
Branch: main
Hash: b38c42262f58cf0587b627e7a866d3756da34dc4
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

Roy-043, 2024-09-30:
I have to reopen this issue as it has not been fixed.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38843 (Git)
Build type: Release
Branch: main
Hash: d519f24df9931e4a98d19ee39cdcd6124ae6a7bd
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods:
```

davesrocketshop, 2024-09-30:
Yes. The python custom attribute did not make the change. This should be a quick fix.

davesrocketshop, 2024-10-01:
Test cases all pass. Adding them was quite helpful

closed 2024-10-03 by chennes

leoheck, 2024-11-24:
This may not be fixed fine.
I am experiencing something related to Transparency and Diffuse colors. 
But it is tough to reproduce more than twice in a row.
https://github.com/FreeCAD/FreeCAD/issues/18075

PR #16692 (2024-09-21): Materials: Set transparency from DiffuseColor
Migrate the transparency from the DiffuseColor alpha channel to the transparency color attribute. This allows the alpha channel to be used as designed

fixes #14938
commit 84b0109a (2024-09-21): Materials: Set transparency from DiffuseColor

Migrate the transparency from the DiffuseColor alpha channel to the
transparency color attribute. This allows the alpha channel to be
used as designed

fixes #14938
commit a5c4416b (2024-09-21): Materials: Set transparency from DiffuseColor

Migrate the transparency from the DiffuseColor alpha channel to the
transparency color attribute. This allows the alpha channel to be
used as designed

fixes #14938
PR #16941 (2024-10-03): Materials: Correct DiffuseColor custom attribute
Custom attributes were modified to maintain the behaviour of setting transparencies using the DiffuseColor alpha channels

fixes #14938