FreeCAD/FreeCAD #14939: Material: Cannot set transparency per face in the GUI

Roy-043, 2024-06-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There is no Transparency control in the new dialog of the Part_ColorPerFace command. In the v0.21 version of the command you could specify the alpha channel (opposite of transparency) in the "Select Color" dialog. See attached image (alpha 38 means transparency is 85%).

[image]

### Full version info

```shell
[code]
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
[/code]
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-09-06:
Related PR: https://github.com/FreeCAD/FreeCAD/pull/16318

luzpaz, 2024-10-23:
[image]
https://github.com/FreeCAD/FreeCAD/pull/16318#issuecomment-2333932589

@davesrocketshop this is good to close, correct ?

davesrocketshop, 2024-10-23:
> ![Screenshot_20241023_093903](https://private-user-images.githubusercontent.com/4140247/379294967-188b3f37-5adf-429f-9abf-6c990a88e0b6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk2OTI3OTksIm5iZiI6MTcyOTY5MjQ5OSwicGF0aCI6Ii80MTQwMjQ3LzM3OTI5NDk2Ny0xODhiM2YzNy01YWRmLTQyOWYtOWFiZi02Yzk5MGE4OGUwYjYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAyMyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMjNUMTQwODE5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZDY4ODNjM2Q4MDhkY2Q3ZGJlOTUzZDdjNjhmMTM2NGI5YTg3ZDc2NDIzZWI5YzgxNmRiMTllYWY2YzE5MDczZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.XTs4IplpMcxGMKQjyvtGidIhBSkGHaK-ppBHzlFTECk) [#16318 (comment)](https://github.com/FreeCAD/FreeCAD/pull/16318#issuecomment-2333932589)
> 
> @davesrocketshop this is good to close, correct ?

Yes