FreeCAD/FreeCAD #14943: Error "Removing splitter failed" when generating a threaded hole that's countersunk

Takrel, 2024-06-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I receive the error "Removing splitter failed" when i want to make a hole with visible thread in it that's countersunk.

Steps to reproduce:
1. Make on a face on a body a sketch with a 2.5mm hole.
2. Define it fully so the solver is happy
3. Use the holes tool and make a M3 thread that's visible with 3.5mm countersunk

[image]
[image]
[image]

Example file with 4 holes. As soon as you set them to countersunk you will get the error.
[1-01.1L-0.22.FCStd.zip](https://github.com/user-attachments/files/15972776/1-01.1L-0.22.FCStd.zip)

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37841 (Git) AppImage
Build type: Release
Branch: main
Hash: 5f761c758e2a0164b00a7183b9c0cbe32688e123
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Switzerland (de_CH)
Installed mods: 
  * OpenTheme 2024.6.22
  * fasteners 0.5.24
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Takrel, 2024-07-02:
Found a workaround for it.
Disable refine in the settings of the last feature helps.
I also turned all the refine settings off from the workbench part design.

Takrel, 2024-07-02:
Not closed. Wrong button. Sorry

luzpaz, 2024-07-22:
@FlachyJoe do you have a moment to look at this ?

AIRCAP, 2024-11-12:
cannot reproduce the issue in latest main, can you confirm this is fixed?

maxwxyz, 2024-11-12:
Should be fixed in  https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
Feel free to reopen if there is still an issue

closed 2024-11-12 by maxwxyz

RiccaDS, 2025-04-16:
I have the same "Removing splitter failed" problem on latest dev. 
Note I have disabled recomputes for the document, otherwise I get a blank error message window when I try to apply the thread. 

[thread error.zip](https://github.com/user-attachments/files/19778003/thread.error.zip)

```
OS: Ubuntu 24.10 (ubuntu:GNOME/ubuntu/xcb)
Architecture: x86_64
Version: 1.1.0dev.41232 (Git) Conda AppImage
Build date: 2025/04/11 14:31:02
Build type: Release
Branch: main
Hash: 1983052e381b5e5dbc78778186dc8f85a23163e5
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/140.99/1
Installed mods: 
  * Curves 0.6.61
  * Manipulator 1.5.9
  * Estimate 0.1.2
  * fasteners 0.5.38
  * ThreadProfile 1.93.0
  * Defeaturing 1.2.4
  * A2plus 0.4.68
  * sheetmetal 0.7.22
```

FlachyJoe, 2025-04-16:
Duplicate #20580
See also #17846 #18110