FreeCAD/FreeCAD #14944: Gordon Surfaces get wrecked when close positioned vertices of sketch and Freehand_BSplines are placed near to each other

EarlJay1, 2024-06-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Creating a Gordon surface, it can happen that when a sketch of it is modified, the Freehand_Bsplines that have end points close to each other get snapped into one point, resulting in faulting recalculation of the Gordon. Can't be rescued easily as well. That behaviour wasn't noticed in 0.21.2.
In the example file, open Sketch001 and move the right end point slightly upwards, that triggers the behaviour.
[gordon test.zip](https://github.com/user-attachments/files/15972815/gordon.test.zip)

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37841 (Git) AppImage
Build type: Release
Branch: main
Hash: 5f761c758e2a0164b00a7183b9c0cbe32688e123
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Curves 0.6.39
  * fasteners 0.5.24
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-25:
This issue should be in the CurvesWB repository

closed 2024-06-25 by maxwxyz

EarlJay1, 2024-06-25:
Sorry I am not familiar with this stuff. Where can I find this repo?

EarlJay1, 2024-06-25:
Nvm, think I found it.
Did not consider it as wrong here since the bug occurs in the Freecad version jump from 0.21.2 to current snap and thought it might be related to TNP or something recently changed.