FreeCAD/FreeCAD #14945: Sketcher: crash (segfault) when canceling B-Spline operation

xphil, 2024-06-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Segfault in Sketcher, when creating B-Spline.
Reproducible on all latest FreeCAD/Ondsel releases including weekly builds.

Steps to reproduce:
1. Create new document
2. Switch to Part Design, add new Body and then Sketch (orientation doesn't matter)
3. When in Sketcher, click B-Spline tool
4. Start drawing B-Spline exactly from 0,0 point (doesn't reproduce from other places).
5. Once you started dragging B-Spline from 0,0 in any direction, do a right mouse button click.

It will crash on right click, while dragging B-Spline from 0,0 point.

Other user on FreeCAD Forum reported no crash on Windows, but a bunch of errors instead.

If you need more information - just let me know.

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git) AppImage
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Sketcher

### Anything else?

FreeCAD Forum topic link:
[Sketcher: crash when canceling B-Spline operation](https://forum.freecad.org/viewtopic.php?t=88693)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-06-25:
> Other user on FreeCAD Forum reported no crash on Windows, but a bunch of errors instead.

Yes, as I've mentioned in the forum thread, there's no crash on Windows but the following errors appear:

```
16:20:53  Sketcher constraint number 1 is malformed!
16:20:53  pyException: Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'IndexError'>: Constraint has invalid indexes
16:20:53  Unhandled exception in ViewProvider::eventCallback: Constraint has invalid indexes
(Event type: SoMouseButtonEvent, object type: SketcherGui::ViewProviderSketch)
16:20:53  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
16:20:53  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
16:20:53  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
```

@PaddleStroke It might be caused by #14079.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

PaddleStroke, 2024-06-25:
Yes it is likely 14079. I will check this

PR #14951 (2024-07-08): Sketcher: BSpline: Fix crash when cancelled while only one autoconstrained point
Fixes #14945
closed 2024-07-08 by chennes