FreeCAD/FreeCAD #14947: Feature Change- Expression editor does not show properties after the dot

JDMils, 2024-06-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The expression editor currently works, however it does not work like most programming languages for displaying properties after entering the dot. My Forum topic on this issue is located here: https://forum.freecad.org/viewtopic.php?p=767297#p767297

So, in most programming languages, you would enter the object name then as soon as you press the dot ".", all the properties of the object appear. Once you've selected the property, you can enter a further dot "." so that the system can show you further available properties and it continues like this until you run out of properties to view.

FreeCad does this illogically- when you select the object/property, it automatically adds the dot "." at the end and does not show the next set of available properties which makes selecting the next set of properties difficult.

Here's what happens in FCAD:
* Enter Expression Builder.
* In most cases a dimension is displayed. Clear this using the back key.
* Type "Sk" to find "Sketch_Profile". FC now shows "<<Sketch_Profile>>." (note the dot at the end) but no properties. How do I get a view of the available properties now? If I type a character here, FCAD will show me the available properties containing that character. For example, typing "x" shows the properties "ExternalGeometry" & "ExpressionEngine".
* Blindly, I type "C" and select "Constraints" whereby FCAD now shows "<<Sketch_Profile>>.Constraints." (note the dot at  the end) but no properties. If I select the Constraints property by pressing <down-arrow>, <Enter> then I get the next set of available properties.
* If I type in any random character, like "s" and then backspace it, I see a list of available properties so I choose one and click OK.

IMO, the way Expression Builder should work is this:
* Enter Expression Builder.
* Type "sk" to find "Sketch_Profile". FCAD should show "<<Sketch_Profile>>" (note: no dot on the end).
* Type ".". FCAD should now show all available properties. I select "Constraints". FCAD should now show "<<Sketch_Profile>>.Constraints" (note: no dot on the end).
* * Type ".". FCAD should now show all available properties. I select "z_total". FCAD should now show "<<Sketch_Profile>>.Constraints.z_total".
* Click OK to accept.

My suggested way is the way most programming languages work today so I really hope we can modify FCAD so that it does not do the same type of operations using a non-standard method.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37573 (Git)
Build type: Release
Branch: main
Hash: 91aac05021023c946dc1545f2b13512ccf326550
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-28:
Related: https://github.com/FreeCAD/FreeCAD/issues/12796

JDMils, 2024-12-27:
This is the scenario which I am trying to convey in issue 14947:

Scenario:

* User select to constrain a shape.
* User selects the Expression editor button in the Constraint box.
* FC provides the current dimension of the selected object.
* User types "sketch" and is presented with 2 objects: "Sketch" and "<>".
* User pressed down-arrow on keyboard to select either object.
* FC automatically shows "Sketch." in the editor however no properties are shown. *** This is the issue I want to bring to the developer's attention. FC should now show all relevant properties since the editor textbox now has a an object and "." prefix but does not.
* User must press ENTER at this point to get a list of relevant properties. This is an extra keypress which should not be required.

Another problem is this:

If the user presses ENTER, all relevant properties are shown. However, even after pressing ENTER and all properties are shown, if the ENTER key is pressed a 2nd time, FC no longer shows the relevant properties and no matter how many times you press ENTER now, no relevant properties are shown.
In this case, if ENTER is pressed in the Expression editor text box, if the currently selected object has a "." suffix, ENTER must show all relevant properties.

luzpaz, 2025-01-25:
@maxwxyz there was some recent system-wide implementation of auto-complete ? correct? I can't seem to locate the commit or issue. I wonder if it impacts this issue?

maxwxyz, 2025-01-26:
@luzpaz https://github.com/FreeCAD/FreeCAD/pull/17058