FreeCAD/FreeCAD #14950: Opening file with Draft Array (Link) has major performance degradation

Syres916, 2024-06-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Using test file (30-racks.FCStd) from https://forum.freecad.org/viewtopic.php?p=767830#p767830 which prior to https://github.com/FreeCAD/FreeCAD/commit/2657bbee4d72df816438f9d9afd7ba860ace6001 took just over _2 seconds_ to load now takes _2 minutes 52 seconds_ on my system. I've done some simple investigation and I'm struggling to find which line is responsible or whether there's been a more recent commit that's made the situation more difficult to pin down the root cause.

FAO @bgbsww

Recreating the file as a non-Link Draft Array is quicker at loading, 12 seconds but this is a workaround as it makes the file more than twice the size.

### Full version info

```shell
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37914 (Git)
Build type: Release
Branch: main
Hash: 6fa8e4ccfbb7e7479521b31910d3c4ed3e6b85ea
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Silk 0.1.5
  * SketchArch
  * freecad.gears 1.2.0
  * Curves 0.6.23
  * CfdOF 1.25.11
  * fasteners 0.5.17
  * sheetmetal 0.4.1
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-06-26:
Also please see Werner's feedback https://forum.freecad.org/viewtopic.php?p=767915#p767912

maxwxyz, 2024-06-28:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/14952

closed 2024-06-28 by maxwxyz