FreeCAD/FreeCAD #14953: Theme can not be applied

borish43b, 2024-06-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I am not able to change the theme.

### Steps:

1.     Select "Edit -> Preferences..."
2.     Change "Theme" in the "General::General" section
3.     Click "Apply" or "OK"

### Expected behavior:

- The theme is changes to the selected one

### Observed behavior:
The following error is yielded:
`boost::filesystem::create_directories: Zugriff verweigert [system:5]: "C:\Users\[[USERNAME]]\AppData\Roaming\FreeCAD\SavedPreferencePacks\Backups", "C:\Users\[[USERNAME]]\AppData\Roaming\FreeCAD"`
The theme is NOT applied.

### Additions:

- The selected theme and all other settings are persisted. If I close and restart FreeCAD, the selected theme is shown in the preferences. The theme is not applied, however.
- All other settings (e.g. language) take effect and are persisted as expected.
- [[USERNAME]] contains a "German umlaut" letter (ä). There has been some issues with this in the past, already. Maybe this causes the issue.

This seems to be an issue with the character encoding in the path (again, there had been an issues with this for the plugins a few versions back).
I have created a new user **“Tester”** and everything worked fine.
Then I have created another user **“Täster”** and I got the exception again.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: German/Germany (de_DE) ]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-28:
Could you try this with the stable 0.21.2 version?
I cannot reproduce with both versions.

borish43b, 2024-06-28:
You are right. I have the same issue with 0.21.2. Maybe it's an UAC issue but then, why can everything else be written without issues and why does it only happen if the user name contains an umlaut?

maxwxyz, 2024-06-29:
@MisterMakerNL @kadet1090 do you know this? It seems to work on Windows if the username has no äöü... in it.

borish43b, 2024-07-02:
I have an update on this:
I played around with the env-vars to change the default settings folders a bit:

- FREECAD_USER_HOME
- FREECAD_USER_DATA
- FREECAD_USER_TEMP

I let them all three point to folder `d:\tmp\fc_test` and everything works fine.

Then I let them point to `d:\tmp\Häppo` and got the following (different) exception when I change the theme
[image]
Again, the exception only occurs when the theme is changed but not when other settings are changes. This time however, the theme changes despite the exception.

maxwxyz, 2024-07-02:
Maybe related: https://github.com/FreeCAD/FreeCAD/issues/11046

borish43b, 2024-08-01:
I have changed the default config directory (my home) only for testing. The issue is affecting the default directory (still in the latest build) because it contains an umlaut.

closed 2025-07-22 by github-actions[bot]