FreeCAD/FreeCAD #14956: TTF fonts with ligatures are not rendered correctly in Draft Shape From Text

hgroover, 2024-06-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

See below for steps to reproduce (which includes using a specific open-source ttf font).
Fonts for composited script languages (such as Sanskrit and other Indic languages) often use ligature glyphs which may come BEFORE other glyphs. The example given here is Kanchi (part of a place-name, Kanchipuram in Tamil Nadu). The Shape From Text feature in the Draft workbench will render this incorrectly, with the "i" ligature placed after the nc conjunct. 
The UTF-8 text being pasted is काञ्चि (unicode) which is e0 a4 95 e0 a4 be e0 a4 9e e0 a5 8d e0 a4 9a e0 a4 bf  in hex. Using IAST transliteration it is kāñci.
The UTF-8 glyphs (and their Unicode equivalents) are
e0 a4 95 - k glyph (devanagari letter ka) U+0915
e0 a4 be - long a modifier (postposition) ligature glyph (Devanagari vowel sign Aa) U+093E
e0 a4 9e - Devanagari na, U+091E
e0 a5 8d - nc conjunct, U+E0AA
e0 a4 9a - Devanagari ca, U+091A
e0 a4 bf - Devanagari vowel sign I (preposition) U+093F

You can see that the UTF-8 string renders correctly in the UI but the Draft conversion to a shape does not get the positioning right.
[image]

An even simpler example that has no consonant conjuncts is काचि

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Anything else?

To reproduce this you'll need the free Sanskrit 2003 font, attached here.
[Sanskrit2003.zip](https://github.com/user-attachments/files/15987041/Sanskrit2003.zip)
Go to Draft, Shape from Text, select the Sanskrit2003.ttf, and paste in

 काञ्चि 

as the text.

You **might** first need to install Sanskrit 2003 as a system font. However, this should be reproducible with the text

काचि

using the devanagari glyphs in one of the fonts such as Arial that has a more complete coverage of the Unicode ranges.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

hgroover, 2024-06-26:
A workaround for this exists. The workaround is as follows:
1. Go to Inkscape, create a text object in the default layer in the desired font, and size it appropriately.
2. In Inkscape, convert the text object to paths.
3. Export as dxf (save as)
4. Import the dxf into FreeCad
5. Switch to Draft workbench, select all the parts imported, and use Convert Draft to Sketch.
6. Switch to Part workbench and extrude the sketch

Roy-043, 2024-12-04:
New development: #16928.

Roy-043, 2025-07-01:
Another development: #22217.

FEA-eng, 2026-01-08:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.2.0dev.20260106 (Git shallow)
Build date: 2026/01/06 15:36:19
Build type: Release
Branch: (HEAD detached at 9b64da8)
Hash: 9b64da827a112d88a025be26316e3d023ff491dc
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```