FreeCAD/FreeCAD #14961: Problems with the new Insert View Button with large 3D files

Libi91, 2024-06-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I use TechDraw to convert large 3D files into 2D isometric views. I started the process with the old Insert View button as soon as I had the graphic in the desired isometric view. Now the window opens and FreeCAD is already working on a front view. This is very annoying with large files because it can take 30 minutes until the front view is created and you can select an isometric view. I also cannot find a way to preselect a camera view in the settings. This was better in the old versions when there was a separate button for a camera view.

### Full version info

```shell
OS: macOS 13.6.4
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * 3DfindIT 1.2.0
  * Part-o-magic 1.1.0
  * freecad.gears 1.0.0
  * InventorLoader 1.4.0
  * Render 2024.1.23
  * pcb 6.2023.0
  * A2plus 0.4.60n
  * Manipulator 1.5.7
  * fasteners 0.4.56
  * CurvedShapes 1.0.4
  * sheetmetal 0.2.63
  * CubeMenu
  * ExplodedAssembly
  * Curves 0.6.12
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-06-27:
I know this is only a workaround but I knocked together a bit of Python which can help out in the meantime if you save it as a macro (obviously you can remove/change the scale accordingly, I was testing with a 5 metre high assembly) and assumes Page already exists:

```
sel = Gui.Selection.getSelection()
if not sel:
    App.Console.PrintWarning("Select something first!\n\n")
else:
    for obj in sel:
        newView = App.activeDocument().addObject('TechDraw::DrawProjGroupItem')
        newView.Source = obj
        newView.Direction = FreeCAD.Vector(0.58, -0.58, 0.58)
        newView.RotationVector = FreeCAD.Vector(1.0, 0.0, 0.0)
        newView.XDirection = FreeCAD.Vector(0.71, 0.71, 0.0)
        newView.ScaleType = "Custom"
        newView.Scale = 0.100000
        App.activeDocument().Page.addView(newView)
        App.activeDocument().recompute(None,True,True)
Gui.SendMsgToActiveView("ViewFit")
```

maxwxyz, 2024-06-28:
@PaddleStroke FYI

benj5378, 2024-08-28:
Maybe the solution here is, that the default orientation is the camera view. That could potentially solve the issue?

PaddleStroke, 2024-08-28:
Yes we can just change the default to use the camera orientation as default.
Having said that, the fact that it can take 30 minutes to compute a view is maybe a problem by itself.

WandererFan, 2024-10-19:
[image]

closed 2024-10-19 by WandererFan

Libi91, 2024-10-19:
I tested "Use 3d Camara Direction". Unfortunately, I don't always get the desired result, like with the old button in version 0.21.
Within a file, some parts are shown in the "Camera Direction", others are not.
Unfortunately, this doesn't solve the problem for me.

benj5378, 2024-10-19:
Maybe add a `Use 3d Camara Direction` to the add view task