FreeCAD/FreeCAD #14963: Unable to import FreeCAD in python script

jamincollins, 2024-06-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This issue was originally reported here: https://github.com/FreeCAD/FreeCAD-Bundle/issues/227
A similar older report can be found here: https://github.com/FreeCAD/FreeCAD/issues/9804

```$ python3
Python 3.12.4 (main, Jun  7 2024, 06:33:07) [GCC 14.1.1 20240522] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys
>>> sys.path.append("/usr/lib/freecad/lib")
>>> import FreeCAD
Fatal Python error: PyImport_AppendInittab: PyImport_AppendInittab() may not be called after Py_Initialize()
Python runtime state: initialized

Current thread 0x0000773c6644d740 (most recent call first):
  File "<frozen importlib._bootstrap>", line 488 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 1289 in create_module
  File "<frozen importlib._bootstrap>", line 813 in module_from_spec
  File "<frozen importlib._bootstrap>", line 921 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1331 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1360 in _find_and_load
  File "<stdin>", line 1 in <module>
Aborted (core dumped)
```

### Full version info

```shell
[code]
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.12.3, Qt 6.7.0, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * CfdOF 1.25.2
  * A2plus 0.4.64
  * Assembly4 0.50.9
  * Assembly3 0.12.2
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

bdieterm, 2024-06-26:
This is already fixed for FreeCAD 0.22 dev
see https://github.com/FreeCAD/FreeCAD/pull/14449

jamincollins, 2024-06-28:
Thank you, the referenced patch resolves my issue.

closed 2024-06-28 by jamincollins