FreeCAD/FreeCAD #14964: Sketcher: new BSpline tool: crash when selecting mode by knots

Roy-043, 2024-06-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If I change the mode of the new BSpline tool to "By knots" there is an instant crash.

Related PR: #14079.

Steps:
1. Start a document.
2. Switch to the SKecther WB.
3. Start a new sketch.
4. Click on the Sketcher_CreateBSpline button.
5. Select "By knots" in the "Mode" dropdown.
6. Result: crash.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-06-26:
2 additional finding:

A.
1. Gui.runCommand('Sketcher_CreateBSplineByInterpolation',0)
2. Switch to "By control points".
3. Error message:
```
22:18:12  Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: Access violation
```

B.
1. Gui.runCommand('Sketcher_CompCreateBSpline',2)
2. Switch to "By control points".
3. Error message:
```
22:18:49  Unhandled std::exception caught in GUIApplication::notify.
The error message is: Access violation - no RTTI data!
```

In both cases the newly created sketch is gone.

PaddleStroke, 2024-06-27:
So it appears that this crash happens only if you just created the sketch.
So if you do : 

- Start a document.
- Switch to the Sketcher WB.
- Start a new sketch.
- Close sketch and reopen.
- Click on the Sketcher_CreateBSpline button.
- Select "By knots" in the "Mode" dropdown.
- Result: crash.

Then it does not crash.

 Which is why I've not detected it during dev as I was using a test document that already had a sketch.
I will investigate.

Roy-043, 2024-06-27:
> Then it does not crash.

Confirmed.

The crash also does not occur if you first create a B-spline by control points.

closed 2024-07-08 by chennes