FreeCAD/FreeCAD #14965: Sketcher: appearance of 3D models should be improved

FEA-eng, 2024-06-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently, 3D models are displayed like this when sketching:

[image]

I know that #14386 was introduced to fix a very annoying issue not fully resolved by #14013 (at least in non-default themes) but IMO the current appearance really needs some improvement as it was much better before the introduction of headlight.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-27:
you can deactivate this in the preferences.
@FreeCAD/design-working-group should this be deactivated by default?

FEA-eng, 2024-06-27:
> you can deactivate this in the preferences. @FreeCAD/design-working-group should this be deactivated by default?

Yeah, but then the issue with default white sketcher geometry being invisible on the faces of the existing geometry is back, right? I would leave it enabled by default, this simplified rendering is much better than not seeing the geometry at all. Perhaps this just needs a different solution in the future, like changing something within the headlight.

obelisk79, 2024-06-27:
Keeping it enabled by default is a definite improvement and should remain. Whether or not there is a better solution to think about moving forward I am unsure. It's a challenging problem.

maxwxyz, 2024-06-28:
I guess this is redundant if https://github.com/FreeCAD/FreeCAD/issues/12824 is solved.
Also related but solved with above issue is https://github.com/FreeCAD/FreeCAD/issues/14967
Should we close this or keep it open for another intermediate fixing attempt?

FEA-eng, 2024-06-28:
> I guess this is redundant if #12824 is solved. Also related but solved with above issue is #14967 Should we close this or keep it open for another intermediate fixing attempt?

I think that this one should stay open because improving the appearance of 3D models in general (a much bigger effort) is different than just finding a better way to fix this nasty issue with the headlight. Most likely this issue will be closed way before the other one.

pierreporte, 2024-06-28:
I have been playing with the headlight direction. It appears that the origin problem disappears when the direction is not normal to the screen or quasi-normal (as it is now). However changing this parameter makes the display overall worse. I think that a good solution would be to implement 3-point lighting, with carefully selected default intensities and direction. It could even improve the overall quality of rendering discussed in #12824 and allow for more realistic materials.

FEA-eng, 2024-07-03:
Not only appearance is affected but there's also a negative impact on the visibility of section cuts: https://forum.freecad.org/viewtopic.php?t=88881#p769145

maxwxyz, 2024-07-03:
@pierreporte @FEA-eng I've commented it here as demo: https://github.com/FreeCAD/FreeCAD/issues/11166#issuecomment-1828460603

@FreeCAD/design-working-group I'm unsure what should be the best, keeping it on by default (fixes parallel plane issue) or disabling it by default (fixes the new issue it introduced: https://forum.freecad.org/viewtopic.php?t=88881#p769145)
Either way this should be addressed soon, either by better light sources (https://github.com/FreeCAD/FreeCAD/issues/11166) or a bigger rendering overhaul (https://github.com/FreeCAD/FreeCAD/issues/12824)

pierreporte, 2024-07-03:
So I see that Catia has only one light source like FreeCAD, and it’s the ambient light that removes shadows. If 3-point lighting is already doable with Python, it may be the easiest to implement. With background illumination on top of this, the issue could be solved with a pretty good result (more realistic appearance of materials). The rendering overhaul could then safely wait.

obelisk79, 2024-07-03:
The shader lacks ambient occlusion lighting, I think this may help in addressing the root of the original problem this had tried to address. Coin3D does support GLSL shaders but still would require someone with the skillset to get implemented. Perhaps offering a paid bounty to the community of shader developers in the minecraft community could be a good avenue to explore. It is a community full of skilled amateurs.

pierreporte, 2024-07-03:
I don’t think that ambient occlusion is part of the solution. In Alibre, which lets users chose whether they want it enabled or not, there is no problem either way with appearance of 3D models. On the other hand, ambient occlusion needs resources that would be better used by the actual design.

obelisk79, 2024-07-03:
AO using OpenGL is going to load the GPU. None of the compute intensive bottlenecks are GPU related so AO would not be a limiting factor unless someone is running FreeCAD on a potato with drivers written on papyrus by an ancient egyptian scribe :)

obelisk79, 2024-07-03:
AO is used in games with highly complex scenes with infinitely more complex geometric meshes going way back to the hardware of the late-1990's. Catia may actually have GPU acceleration built into it, however things like assemblies are almost always CPU limited, same as the Geometric Modeling Kernel, for example OCCT is largely single threaded and doesn't leverage any processing power from a GPU as implemented in FreeCAD. GFX Shaders are processed on the GPU so this is merely a division of labor, and a good AO shader will be quite performant.

luzpaz, 2024-07-07:
Any suggestions on how to make the title of the ticket less generic ?

closed 2025-02-24 by chennes