FreeCAD/FreeCAD #14966: Sketcher: Dimension tool sometimes unnecessarily enters construction mode

FEA-eng, 2024-06-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm not sure what triggers the construction mode of the Dimension tool (I thought it's enabled automatically only when overconstraint would occur but that's not the case). Here it only depends on the location of the label:

[image]

I don't think it's supposed to work like this, right?

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-06-26:
I would guess it's always been like that since OVP was introduced, it certainly was the case in Jan 2024 build and no that behaviour isn't what I would expect, the cursor position shouldn't change a constraint from driving to driven and vica versa IMHO.

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35639 (Git) AppImage
Build type: Release
Branch: main
Hash: f315314569aee5334fd0e69a49aab33684a802ba
Python 3.10.13, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Silk 0.1.5
  * SketchArch
  * freecad.gears 1.2.0
  * Curves 0.6.23
  * CfdOF 1.25.11
  * fasteners 0.5.17
  * sheetmetal 0.4.1
```

FEA-eng, 2024-06-27:
@PaddleStroke

PaddleStroke, 2024-06-27:
This is definitely a bug.
What happens is that the tool switch between Distance and DistanceY when you move the mouse. And somehow for distance its setting the constraint as driving.
I will check what's going on.

closed 2024-07-08 by chennes