FreeCAD/FreeCAD #14967: Highlight of selected faces poorly visible when camera is aligned

Rexbas, 2024-06-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When a face is preselected or selected, the face gets another color. However, when the camera is aligned or almost aligned with the face normal then the highlight color is not or poorly visible. In the video below, I select the face which becomes highlighted green. Then I rotate the camera and you can see that the green becomes lighter and eventually there is no green highlight left. 

The selection should be clearly highlighted regardless of the camera orientation.

### Full version info

```shell
OS: Fedora Linux 40 (KDE Plasma) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37915 +2 (Git)
Build type: Release
Branch: alignment
Hash: c5fd126f4a16a37be5036b38e370198f2ed8c244
Python 3.11.0, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * OpenTheme 2024.6.22
```

### Subproject(s) affected?

None

### Anything else?

https://github.com/FreeCAD/FreeCAD/assets/14298143/9ef636b5-1f0e-4118-a8ae-36c74784f987

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-06-26:
Is this the same as https://forum.freecad.org/viewtopic.php?t=85545?

If so, there is also a potential workaround until this is fixed: https://forum.freecad.org/viewtopic.php?p=754768#p754768

Rexbas, 2024-06-26:
Yes, I think it is the same problem. Changing the `HeadlightDirection` parameter works as a workaround.

wwmayer, 2024-06-27:
As already said several times this is the change to blame: https://forum.freecad.org/viewtopic.php?p=744117#p744117

furgo16, 2024-06-27:
There are 5 pages of forum discussion, and the cause seems to be clear, but not the solution (at least to me, apologies if I'm missing something). This is not a criticism at all, though.

I'm mentioning this to highlight that this will most probably be a source of confusion for at least users who upgrade and users who edit pre-1.0 models. Thus it might be worth targetting a fix or migration mechanism for 1.0.

maxwxyz, 2024-06-27:
related: https://github.com/FreeCAD/FreeCAD/issues/12824
IMO there is no cause as this was always the case when you've used Materials with specularity/shininess and ambient color for better representation. 
A solution is to either
- add a general background illumination in addition to a headlight which is not in line with the camera direction or
- as default a 2 or 3 light setup with no light in the direction of the camera view.

wwmayer, 2024-06-27:
What do you mean with 'always'? The possibility to change permanently the four color channels of a material exits for just two months (with the old appearance dialog it was possible to change it temporarily) and before that only the diffuse color could be changed with the GUI.

At least you can change the default specular color in the preferences  under Part/Part Design > Shape appearance.

maxwxyz, 2024-06-27:
yes always since implementing this.
The changing of the specular color in the preferences is also not ideal IMO as now every setup / theme defines it's own default material instead of having the same default material in main as discussed on Discord.

GeneGH, 2024-06-27:
The proper solution to this entire mess is to allow users to completely opt out of the new materials system. Should be a simple preference toggle.

I never, ever need custom materials or colors. I don't use custom themes, although it seems that everyone is now forced to choose something.

The long-standing default materials behavior worked fine for me, and apparently many others.

pierreporte, 2024-06-27:
@maxwxyz I just opened #14992 for the theme-dependent default material.

krushia, 2024-06-29:
This is #10614

maxwxyz, 2024-06-29:
Duplicate of #10614

closed 2024-06-29 by maxwxyz