FreeCAD/FreeCAD #14971: draft shape 2d view generates extra lines

brettowe, 2024-06-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

a friend was having issues with laser cut files generated from a model made in freecad.
it was narrowed down to having double lines in places.
looking at the model file I found that for some reason when using shape 2d view if the sketch generating say a pocket is resting on the view/face you want to export for cutting.
it will generate extra lines. it looks like it is including the sketches lines and then the pocket's cut lines.
hovering over the shape 2d lines can show the double issue without exporting via autodesk dxf and importing in inkscape
[image]

if you raise the sketch up (sketch attachment z position +1mm) it goes away in the attached example file.
[image]

I noticed it with the ellipse and the original files b-splines, not sure if its limited to those or just shows up easier.
in the original file I also saw one other item that I couldn't reproduce in the example was face selection vs. object (pocket) selection you could still see the double lines on the object selection even with the z position change I now can't get it to reproduce that issue in the original file so there may be another related bug hiding in there also.

bug was found in v0.21and also tested in weekly (version info provided for that)

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.12.3
```

### Subproject(s) affected?

None

### Anything else?

[example-bug-model.zip](https://github.com/user-attachments/files/15995061/example-bug-model.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-06-26:
In part the issue is caused by minute inaccuracies. If I attach Sketch001 (used for the Pocket) to the XY plane of the body (instead of to a face of the Pad) and reverse the Pocket, the problem goes away. But the problem is also related to the shape. If I open the original file and replace the ellipse in Sketch001 with a circle, the problem does not occur.

Remove the zip extension from the attached file to open it.
[example-bug-accuracy.FCStd.zip](https://github.com/user-attachments/files/15995994/example-bug-accuracy.FCStd.zip)

brettowe, 2024-06-26:
I tested the sketch attachment to xz plane and reverse pocket in the original file and it still shows the problem.
and i forgot to mention the bulk of the original file was mainly b-splines and it had a repeating pattern that was cloned and rotated around in the sketch rather than say polar pattern of the pocket.
in the case of that item not all of the clones show the problem its 1/3rd to 1/2

julianjc84, 2024-11-17:
I am also experiencing this issue as well.
happens in freecad 0.21 and upwards 1.0rc4, i have not gone further back.

I am not sure how exclusive to the draft workbench it is.

I have documented some of my findings here [https://forum.freecad.org/viewtopic.php?t=92096](url)

I have also discovered it is only related to bsplines. Circles and squares and other wires are not effected.
only happens when a bspline is applied as a 2nd feature onwards to a existing pad/pocket.

Can see more in the attached sample document.
if you DRAFT:Downgrade the 3 sample shape2dview its very obvious the extra lines that are generated on the 2 part Pad/Pocket body where as the other 2 bodies have no overlap lines.

also note when viewing a extrude or pad bspline under a VIEW:object style:Deviation 0.5, You can seethe non uniform surface. This is only a visual feedback however seems related to what's happening with bsplines in the backend maybe? 
or even applying a curves/SurfaceAnalysis.

rename file without the .zip its not compressed.
[bspline3situtaitnsdowngradetheshape2dview.FCStd.zip](https://github.com/user-attachments/files/17790207/bspline3situtaitnsdowngradetheshape2dview.FCStd.zip)

FEA-eng, 2025-09-29:
Still relevant. I get one ellipse and 5 overlapping splines in a sketch created from Shape2DView object.

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

WandererFan, 2026-03-15:
@Roy-043 projectEx produces a py list containing 10 compounds (some of which might be null) of edges.  There is a  C++ routine that will replace overlapping edges with (possibly multiple) non-overlapping edges:
```
//origEdges should be a copy of the original edges since the underlying TShape will be modified.
std::vector<TopoDS_Edge> DrawProjectSplit::scrubEdges(std::vector<TopoDS_Edge>& origEdges,
                                                      std::vector<TopoDS_Edge> &closedEdges)
```

Unfortunately, scrubEdges is not yet exposed to python.   

A potential issue is that going through scrubEdges would lose the classification information contained in the 10 compounds (visible hard edges, visible outline edges, etc).

EdgeWalker's purpose is to create closed regions from a set of edges.   It does call scrubEdges before attempting to form wires from the edges.  If building the closed wires is not relevant for Shape2DView, then EdgeWalker's graph traversal process (somewhat expensive) is a unneeded step.

closed 2026-09-07 by Roy-043