FreeCAD/FreeCAD #14972: Tree View problems

Kuzma30, 2024-06-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. When Description and Internal Name columns hide and you select by mouse, you get different selected region width
[image]
2. When additional column is shown selecting looks normal (Only highlight first column for Active Body is normal? )
[image]
[image]
3. When Internal names is shown same situation as 1.
[image]
4. Resizable column active and Internal names active
[image]

@kadet1090 

### Full version info

```shell
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37854 (Git)
Build type: Debug
Branch: master
Hash: 4c3fc869feadfd8cc9feab0588d5e5b6c981e624
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.8.2.dev
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 
  * sheetmetal 0.4.18
  * ose-piping
  * dodo 1.0.1
  * fasteners 0.5.22
  * FeedsAndSpeeds 0.5.0
  * ose-3d-printer-workbench 0.1.0
  * OpenTheme 2024.5.16
  * Assembly3 0.12.2
  * addFC 0.3.1
  * ose-workbench-core
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kadet1090, 2024-06-26:
@maxwxyz can you assign me?

luzpaz, 2024-07-27:
Hi @kadet1090, is a fix planned pre- or post-1.0 ?

kadet1090, 2024-07-27:
pre-1.0 - I should be able to submit PR today

PR #15652 (2024-07-29): Gui: Cosmetic Fixes
This PR fixes various small cosmetic problems with preferences, like wrong padding, inconsistent alignment, horizontal scrollbar that should not be present etc.

Examples of before and after:
| Before | After |
| -- | -- |
|[image] | [image] |
| [image] | [image] |
| [image] |[image] |
| [image] | [image] |

Fixes: #14972 
Fixes: #14760 
Fixes: #15308