FreeCAD/FreeCAD #14973: sketch add text

Vincenzoferrara, 2024-06-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I know you can insert text with draft, but I prefer sketch. but I don't see any way to do it.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git) AppImage
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Installed mods: 
  * AutoOpenSpreadsheet
  * IconThemes
  * fasteners.backup1715436808.37841 0.5.16 (Disabled)
  * btl 0.9.9
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.5.16
  * Assembly4.backup1718498214.0742533 0.50.12 (Disabled)
  * fasteners 0.5.24
  * TabBar
  * Assembly4 0.50.13
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-06-27:
Duplicate of #12944

maxwxyz, 2024-06-27:
@Vincenzoferrara please consider
- searching for existing issues before opening new ones
- writing better issues, describing a concise problem and fill out all fields

as in the past I've had to close issues from you.

closed 2024-06-27 by maxwxyz