FreeCAD/FreeCAD #14975: Attachment position of ShapeString not relative to edge bounding box.

theosib, 2024-06-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm trying to auto-center a ShapeString, so the formula for centering it should be something like `Box.Shape.BoundBox.XLength / 2 - .Shape.BoundBox.XLength / 2` or `-.Shape.BoundBox.XLength / 2`, depending on what it's relative to.

Please have a look at the attached file and the ShapeString named "6". I'm trying to use a formula to center it horizontally, but nothing I do works. It's always off center. Clearly FreeCAD knows what the border of the character is. You can see that in the attached image where a green box is snug around the "6". But either FreeCAD is computing the width of the box wrong, or it's not horizontally positioning the ShapeString relative to the left edge of the bounding box.

[die-cargo-glitch.FCStd.zip](https://github.com/user-attachments/files/15997565/die-cargo-glitch.FCStd.zip)
[image]

Note: The font can be found here: https://www.dafont.com/cyber-freight.font

### Full version info

```shell
[code]
OS: macOS 13.6
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

theosib, 2024-06-26:
I used the measurement tool on the 6, and the computed width of the 6 seems to be correct. So it must be that the ShapeString is positioning relative to something other than the left edge of the bounding box. The 6 is attached to the face of the cube. If I set the X coordinate of the 6 to zero, it's neither at the origin nor at origin. 
[image]

I also tried attaching ShapeString to the XY plane of a Std_Part and set the attachment position to the origin. The Y position is correct, but the X position is shifted to the right by some weird amount.

[image]

And like I say, clearly the ShapeString knows where the left edge of the 6 is. The green box is snug around it. So the ShapeString's positive is relative to some unknown and invisible other offset.

maxwxyz, 2024-06-27:
@Roy-043 any idea?

Roy-043, 2024-06-27:
That left margin is the result of how characters are defined in the TTF file. In the current dev version there is a "Keep Left Margin" property. Setting it to false will remove the margin.

In V0.21 you can use `.Shape.BoundBox.XMin` to retrieve the size of the left margin (assuming the object is on the XY plane and aligned to the X axis of course).

Closing this as resolved.

closed 2024-06-27 by Roy-043