FreeCAD/FreeCAD #14976: ShapeString .Shape.BoundBox.XLength is zero for ShapeStrings positioned along YZ plane

theosib, 2024-06-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm trying to auto-position a ShapeString to center it on a face. Please see https://github.com/FreeCAD/FreeCAD/issues/14975 for context. Someone told me that the issue I reported there is fixed in the dev versions, so I'm trying that. But I'm finding a new problem.

Please see the attached file. I'm trying to use formulas like `(Box.Shape.BoundBox.XLength - .Shape.BoundBox.XLength) / 2` to automatically center ShapeStrings relative to an attachment face. This works for ShapeStrings on the top, bottom, front, and rear faces. But left and right faces (YX plane), the XLength for the ShapeString is zero. You can see this for the ShapeStrings named "5" and "2" in the attached file. 

[die-cargo-glitch 2.FCStd.zip](https://github.com/user-attachments/files/15998346/die-cargo-glitch.2.FCStd.zip)
[image]

Note: You can find this font here: https://www.dafont.com/cyber-freight.font

### Full version info

```shell
OS: macOS 13.6
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
```

### Subproject(s) affected?

Draft

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

theosib, 2024-06-27:
Here are some steps to reproduce this minimally:
1. Create an empty document
2. In Draft > Utilities > Select Plane, choose Top (XY)
3. Create an unattached ShapeString. Put a "2" in it.
4. Set the attachment X position to `.Shape.BoundBox.XLength` (just to get access to the value)
5. Notice that the length is computed correctly.
6. Create a Std_Part.
7. Attach the ShapeString to the YZ plane of the Std_Part.
8. Notice that the Base placement value for X has gone to zero.
9. Also notice that settings the Attachment placement to `.Shape.BoundBox.XLength` also produces zero.

Another way to reproduce it:
1. Create an empty document
2. In Draft > Utilities > Select Plane, choose Side (YZ)
3. Create a ShapeString
4. Set the attachment X position to `.Shape.BoundBox.XLength` (just to get access to the value)
5. Notice that the length is computed as zero.

If I attach the ShapeString to XY instead, the length is computed correctly.

theosib, 2024-06-27:
I checked, and when attaching the ShapeString to the YZ plane, ZLength gets the width of the string. This should not be happening.

This is entirely inconsistent with sketches, which makes it even more confusing.

A sketch represents an XY coordinate system, regardless of what it's attached to, and any extrusion is always performed normal to the attachment surface, with the extrusion occurring along its relative Z axis.

For consistency, the ShapeString **should also** represent an internal XY coordinate system, and it should remain internally an XY coordinate system, regardless of what it's attached to. Extrusions of ShapeStrings are **also** normal to the surface, and that normal should be the relative Z axis.

theosib, 2024-06-27:
Now consider the following scenario. Let's say I want to center a ShapeString along a plane whose normal is proportional to <1,1,0>. In that case, the height of the string is still YLength, but to get the width, we have to compute `sqrt(XLength**2 + ZLength**2)`. A successful workaround for a problem that shouldn't be a problem.

But now consider a ShapeString attached to a surface with normal proportional to <1,1,1>. In this case, the width and height of the string's bounding box become unrecoverable.

Roy-043, 2024-06-27:
The BoundBox of objects is always relative to the parent space. That is just how it works in FreeCAD. You can nest your object in a Std_Part, and manipulate the Placement of that object to see if that works for you.

In the dev version there are 9 alignment options for ShapeStrings. Your calculations should not be necessary.
See: https://wiki.freecad.org/Draft_ShapeString

I am closing this.

closed 2024-06-27 by Roy-043