FreeCAD/FreeCAD #14981: Sketcher: new BSpline tool: Degree setting is buggy and confusing

Roy-043, 2024-06-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There are several issues with the degree input box of the new BSpline tool:
1. Changing the value with the spin buttons results in a decimal number instead of an integer.
2. After finishing a spline, zero is always displayed. Clicking the 1st point of a new spline or switching to a different application will update the input box to the correct value.
3. B-splines by knots are always initially created with a degree of 3. Choosing that mode should probably disable the degree input box.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git)
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-06-27:
1 is a problem of all the DSH. It was reported for Polygone already. At the time I couldn't find how to fix this.
2 and 3 are fixed by 14985

Roy-043, 2024-07-02:
Regarding problem 1:
As I understand it all inputs are quantity spinboxes. And there are 10 of those that are used for the OVP parameters as well as in the task panel. Wouldn't it be possible to add a couple of 'normal' spinboxes to that set of 10?

closed 2024-07-08 by chennes