FreeCAD/FreeCAD #14987: Freecad does not honor the property `Allow duplicate object labels in one Document` when pasting a `'Part::Feature'`

leoheck, 2024-06-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Freecad does not honor the property `Allow duplicate object labels in one Document` when pasting a `'Part::Feature'`.
It is still adding `000, 001, 002` at the end of the new object.

### Full version info

```shell
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37898 (Git) AppImage
Build type: Release
Branch: main
Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

7AXEL, 2024-06-27:
Check out this project: https://github.com/7AXEL/Web2APK

leoheck, 2024-06-28:
This was observed when using the unfold feature of SheetMetal. It may be something generated on that workbench but I remember that Freecad can also rename (relabel) objects when there are other existing labels, then I thought it would be ok to report it here too. 

Yesterday, I put a screenshot of this on this comment here:
https://github.com/shaise/FreeCAD_SheetMetal/issues/350#issuecomment-2194810945

Roy-043, 2024-07-03:
Confirmed (the word "pasting" should be "creating").

To verify, paste this multiple times in the Python console:
`FreeCAD.ActiveDocument.addObject("Part::Feature", "Unfold")`

leoheck, 2024-07-03:
Indeed, thanks for the correction.

luzpaz, 2024-07-31:
> To verify, paste this multiple times in the Python console:
FreeCAD.ActiveDocument.addObject("Part::Feature", "Unfold")

@wwmayer JFYI, this is how to reproduce

leoheck, 2024-07-31:
Actually I think I understand now that when generating... stuff that has the same Label receive an incremental/different Label. It doesn't mean that new/generated stuff should not have a unique Label. The property means the user can keep objects with the same Label. Considering this, I understand that there is no issue.