FreeCAD/FreeCAD #14990: Assembly: Insert Component shows items hidden in the Tree view

furgo16, 2024-06-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps to reproduce:

1. Open attached file
2. Notice it has an `Intermediate artifacts` group that is not shown in the Tree view. It has been marked as hidden with the right-click context menu action <kbd>Toggle visibility in the Tree view</kbd>. The components on this group are not meant to be used or seen in the assembly.
3. Click on the <kbd>Insert component</kbd> (`Assembly_InsertLink`) toolbar button
4. Expected: only groups and components visible in the Tree view are shown in the Tasks dialog
5. Actual: the `Intermediate` artifacts group and its components are shown, although they're marked as not visible in the Tree View.

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37915 (Git)
Build type: Release
Branch: main
Hash: 8d3f07c1f73ef5b1829760db251b9524094aeea2
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Installed mods: 
  * OpenTheme 2024.5.16
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-06-28:
The tree view is not the same as the add components tool. I sure want all available components to be visible when adding from existing files.

furgo16, 2024-06-28:
The use case in this issue is for non-reusable intermediate design artifacts that have been explicitly marked as hidden, as they cannot simply be deleted. I thought the motivation for the "Toggle visibility in the tree" switch was to hide those components which are not going to be used, but as a non-experienced CAD designer I might be well missing something.

Ideally, they should not be cluttering the "Insert component" picker, particularly as the tree there is expanded by default. In the example, after clicking on "Insert component" one needs to scroll down past the artifacts every time before being able to pick an assembly component.

[image]

A workaround, I guess, is to wrap all bodies in Parts, except for the artifacts, and then tick the `Show only parts` checkbox.

Or a more flexible solution might be to have a `Show components hidden in the tree` checkbox in the "Insert component" dialog.

PaddleStroke, 2024-07-22:
After thinking about it a bit, I'm not sure it is ideal to introduce a setting for this.
@maxwxyz do you have a usecase in mind where you'd want to insert an object that is currently hidden in the tree view?

Thinking about it, it feels like if the user actively hide an object from the tree view, then he most likely won't need to insert it.

So I would rather just hide the same objects in the insert list as in the tree with no checkbox. If user somehow needs to insert an hidden object he can un-hide it.

maxwxyz, 2024-07-22:
I have no use case in mind.

closed 2024-10-25 by chennes