FreeCAD/FreeCAD #14991: Assembly: cannot move components to group - FreeCAD crash

furgo16, 2024-06-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

## Steps to reproduce:

1. Open attached file: [F-02-C_bug.zip](https://github.com/user-attachments/files/16018759/F-02-C_bug.zip)
2. Create a group via the Std_Group command ("Create group" toolbar button)
3. Drag the group to be under the Assembly in the Tree view
4. Drag the `Outer stile left003` component under the Group
5. **Expected**: the component moves into the Group
6. **Actual**: a modal dialog appears, impending the move. Either option crashes FreeCAD (pressing either Yes, No, or Esc) [image]

## Stacktrace:

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7fea5c5f5520]
#1  0x7fea5d105876 in QDrag::dragCursor(Qt::DropAction) const from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x26
#2  0x7fea5d1092bf in QBasicDrag::updateCursor(Qt::DropAction) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x5f
#3  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x84475) [0x7fea561d0475]
#4  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x84c70) [0x7fea561d0c70]
#5  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x87c32) [0x7fea561d3c32]
#6  0x7fea5d10857e in QBasicDrag::eventFilter(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x19e
#7  0x7fea5cce1a5e in QCoreApplicationPrivate::sendThroughApplicationEventFilters(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x8e
#8  0x7fea5d7d1750 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xc0
#9  0x7fea5ef14bd8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x98
#10  0x7fea5cce1e3a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#11  0x7fea5d0c8307 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x697
#12  0x7fea5d09da2c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0xac
#13  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x73d6e) [0x7fea561bfd6e]
#14  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x26b) [0x7fea5a8e4d3b]
#15  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xab2b8) [0x7fea5a93a2b8]
#16  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7fea5a8e23e3]
#17  0x7fea5cd3b0b8 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x68
#18  0x7fea5cce075b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#19  0x7fea5d107fb7 in QBasicDrag::drag(QDrag*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x47
#20  0x7fea5d1053e9 in QDragManager::drag(QDrag*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x129
#21  0x7fea5d10573f in QDrag::exec(QFlags<Qt::DropAction>, Qt::DropAction) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x7f
#22  0x7fea5da57570 in QAbstractItemView::startDrag(QFlags<Qt::DropAction>) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x150
#23  0x7fea5f0cba30 in Gui::TreeWidget::startDrag(QFlags<Qt::DropAction>) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x20
#24  0x7fea5da55d66 in QAbstractItemView::mouseMoveEvent(QMouseEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x5a6
#25  0x7fea5d8144ee in QWidget::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x20e
#26  0x7fea5d8c2422 in QFrame::event(QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x22
#27  0x7fea5cce1b9a in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0xaa
#28  0x7fea5d7d1702 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x72
#29  0x7fea5d7d9364 in QApplication::notify(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xa84
#30  0x7fea5ef14bd8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x98
#31  0x7fea5cce1e3a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#32  0x7fea5d7d7e47 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1b7
#33  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1c8d40) [0x7fea5d82dd40]
#34  /lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1cbfd5) [0x7fea5d830fd5]
#35  0x7fea5d7d1713 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x83
#36  0x7fea5ef14bd8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x98
#37  0x7fea5cce1e3a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#38  0x7fea5d0c8307 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0x697
#39  0x7fea5d09da2c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Gui.so.5+0xac
#40  /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5(+0x73d6e) [0x7fea561bfd6e]
#41  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x26b) [0x7fea5a8e4d3b]
#42  /lib/x86_64-linux-gnu/libglib-2.0.so.0(+0xab2b8) [0x7fea5a93a2b8]
#43  /lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x33) [0x7fea5a8e23e3]
#44  0x7fea5cd3b0b8 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x68
#45  0x7fea5cce075b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#46  0x7fea5cce8cf4 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x94
#47  0x7fea5ee9bb3a in Gui::Application::runApplication() from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x8ea
#48  freecad-daily(+0x6d54) [0x556dad620d54]
#49  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7fea5c5dcd90]
#50  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7fea5c5dce40]
#51  freecad-daily(+0x7ea5) [0x556dad621ea5]
```

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37915 (Git)
Build type: Release
Branch: main
Hash: 8d3f07c1f73ef5b1829760db251b9524094aeea2
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Installed mods: 
  * OpenTheme 2024.5.16
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2024-06-28:
Confirmed.

This is caused by showing the modal dialog in ViewProviderAssembly::canDragObject(). If removing the dialog it works correctly.
The problem is that canDragObject() is called by the event handler TreeWidget::dragMoveEvent() and there cannot be a call to show a modal dialog because this messes up Qt's event handling and thus leads to the crash afterwards.

@PaddleStroke

PaddleStroke, 2024-06-28:
Thanks for the report and for the analysis.

It used to work until https://github.com/FreeCAD/FreeCAD/pull/14733.

It's because `canDragObject` was actually never called in `dragMoveEvent `before since this block was using the condition `&& !vp->canDragAndDropObject(obj)` which didn't seem to make any sense so this block was never triggering.

So anyway the problem is that this code making the dialog doesn't seem to belong to canDragObject.

On top of that ideally we should not need this dialog. The joints could be updated automatically based on where the object is dropped. Or deleted if the object is dropped outside of assembly.

PaddleStroke, 2024-06-28:
Current approach is not the best anyway because if the object is within a group (that is in the assembly) then dragging it out of assembly is not caught as its the group canDragObject that is called and not the assembly one.

The problem is that if we just remove this deletion of joints when an object is removed from the assembly, then we get errors : 

> 11:58:06  Assembly::JointGroup: Link(s) to object(s) 'GroundedJoint Revolute' go out of the allowed scope 'Joints'. Instead, the linked object(s) reside within 'N/A'.

And a broken joint.

So we need to find another way to remove joints if the object is removed. Maybe when the joint itself is recomputed, it could check if its object is valid, and if not delete itself.

closed 2024-07-15 by yorikvanhavre