FreeCAD/FreeCAD #14992: Default material is defined by the theme

pierreporte, 2024-06-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The default solid material is hardcoded in files and set by the theme. It means that the theme has an impact. Not only can it create unwanted inconsistencies in project, but the visual appearance of the program has nothing to with materials. The default material should be a program setting and themes shouldn’t be allowed to change it.

### Full version info

```shell
0.22
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

Original issue: #14849.

Discussed with @FreeCAD/design-working-group.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kadet1090, 2024-06-27:
This is more of issue with themes than FreeCAD - themes are essentially preference packs and so they can include any user parameter. I don't think that we should have some settings excluded from preference packs so the best solution seems to simply not include that setting in preferences - @MisterMakerNL @obelisk79

obelisk79, 2024-06-27:
I think the praying on this issue is incorrect. Because, for example, OpenThemes don't set material, merely color. Which imo makes some sense as selecting complementary colors which have good contrast against a given background and set of sketcher colors makes sense. I'm simply just not in favor of taking that away. If anything, a solution would be to give objects which bear the default color a special flag to aid with portability between systems/themes.

pierreporte, 2024-06-27:
I can change the title to “Default material <ins>can be</ins> defined by the theme”.

kadet1090, 2024-06-28:
> I'm simply just not in favor of taking that away. If anything, a solution would be to give objects which bear the default color a special flag that will follow the current default color assigned via preference.

While I do agree with that fully, and I'll implement something like this for next release cycle we should not introduce such inconsistencies in the current release. Contrast will still be an issue because if I'm using light theme and hence have darker default color and I share you a file you'll have much worse experience if you are on dark theme because you'll have much lower contrast. 

The solution should be:
 - For next release let's introduce "Auto Material" / "Auto Solid Color" just like it was introduced for sketcher https://github.com/FreeCAD/FreeCAD/pull/14322 (same thing probably should be done for draft). This way if objects that have not been changed explicitly to have some color will adjust to the current theme./'
 - For this release cycle we should not change the default colors in themes as it can introduce some portability problems between different themes which is not expected by users.

adrianinsaval, 2024-07-01:
> I think the phrasing on this issue is incorrect. Because, for example, OpenThemes doesn't set material, merely default color. Which imo makes some sense as selecting complementary colors which have good contrast against a given background and set of sketcher colors makes sense. I'm simply just not in favor of taking that away. If anything, a solution would be to give objects which bear the default color a special flag that will follow the current default color assigned via preference. This would aid in file portability as well.

but does it really make sense to change default color of objects via a theme? This only makes sense if the model is created and eternally used by the same user and he would have to eternally keep using he same theme

maxwxyz, 2024-07-02:
@davesrocketshop it would be great if you could help on this issue too

obelisk79, 2024-07-30:
> * For this release cycle we should not change the default colors in themes as it can introduce some portability problems between different themes which is not expected by users.

while I can understand the sentiment here, I'd like to point out this issue: https://github.com/obelisk79/OpenTheme/issues/51

FreeCAD Dark theme deals with the same problem by defining very light line/wire colors. This also suffers from the same portability problem. I see the 1.0 label was removed from this issue, but I'd suggest that be reconsidered if kadet is open to submitting a pr. Even without modifying shape color, I need to modify line/edge/wire color for general usability in draft.

@chennes @yorikvanhavre @wwmayer do you have any opinions regarding allowing such a change before 1.0 release despite the current feature freeze status?

chennes, 2024-07-30:
I have no problem with minor tweaks to colors -- I doubt anyone reading tutorials or manuals will be confused between shades of gray, etc. A wholesale color update is overdue, but I don't think that's reasonable at this stage. Start planning for 1.1 :).

kadet1090, 2024-07-30:
IMHO it is not as big of a deal for the material as solids are quite big, have outlines etc so they are quite visible. Draft is another story - we should probably introduce mechanism like one made for sketcher in https://github.com/FreeCAD/FreeCAD/pull/14322 because draft wires are almost invisible and it is usability issue, not a feature.

MisterMakerNL, 2024-11-13:
Same issue is with spreadsheet text color that is currently in a fight with user color settings. 
I couldn't find the correct ticket but this one seems to be similar.
https://github.com/FreeCAD/FreeCAD/issues/16950

maxwxyz, 2024-11-26:
@davesrocketshop is this on your radar for 1.1?

davesrocketshop, 2024-11-26:
> @davesrocketshop is this on your radar for 1.1?

No. This is more for the themes folks. I'll help in any way I can, but I shouldn't be the lead on this

davesrocketshop, 2024-11-26:
> @davesrocketshop I mean there is still the default appearance in the preferences instead of a default material. In my understanding the default material is always overwritten as long as we have these preferences.

There are valid reasons for having it as a preference. Taking it away would in my opinion would be a bad idea and would certainly meet with resistance. A better solution would be some mechanism of preventing themes from overriding certain preferences. That is beyond my scope and not something I'm looking to take on.

davesrocketshop, 2024-11-26:
For example, themes should probably not be overriding default units, There are many other similar preferences that would fall in this category.

maxwxyz, 2025-05-17:
Is this issue also the reason for:
- https://github.com/FreeCAD/FreeCAD/issues/6069
- https://github.com/FreeCAD/FreeCAD/issues/15642
- https://github.com/FreeCAD/FreeCAD/issues/11210

When custom colors are assigned to faces, the next operation destroys the appearance again. Also just creating a new body from a solid, the body and it's base feature override the appearance.

Maybe related:
- https://github.com/FreeCAD/FreeCAD/issues/18262

maxwxyz, 2025-05-17:
not sure what is saved in the file and what is generated locally on each instance depending on the theme and chosen material.

obelisk79, 2025-05-17:
> So the take-away here is that this is purely aesthetic. If a group is collaborating on the same .FCStd file but have different themes...this issues doesn't impact them, right ?

essentially, yes, you are correct.

obelisk79, 2025-05-17:
I'm just going to ping @kadet1090 here as a reminder so perhaps we could see the default color be given an internal alias that is assigned to objects instead of it being a discrete value saved to objects as a if it was manually modified by the user. This should afford improved portability between users with different setups.

MisterMakerNL, 2025-05-18:
With the new theme tokens, I think we can come up with a better solution.
The user has a default material.
But we then add a token called darken(material,10) or lighten(material,10) or whatever is necessary too correct contrast issues with the theme background.
If we apply this correction to every material, we can hopefully prevent any contrast issues.
User will hopefully not really notice the difference, I hope it's not necessary to crank up the darkness or lightness too much.
It's basically a gamma correctionon the material, maybe we can just set a gamma correction instead.🤔

obelisk79, 2025-05-18:
Color theory goes well beyond gamma adjustments. While what you suggest may be viable from a technical standpoint and even maintain aesthetic under certain circumstances, I'm not sure I see the real-world value in doing so. So now the theme could set a color, and tinker with gamma/lightness on top of that etc. I see that as adding unnecessary complexity for minimal gain. But that's just my initial reaction, color theory is a deep well to draw from.

MisterMakerNL, 2025-05-18:
Theme should not change the CAD file in anyway.
Proposed solution will only take the color from the CAD file and change the brightness to make it better suited to the user screen and settings.
It's just a setting that changes the gamma or brightness of a CAD model that is stored in the user settings and is supplied by a theme.
It will not be stored in the CAD file.

maxwxyz, 2025-05-18:
As it currently works, the theme changes the CAD file, as it overrides the appearance properties of the material `Default`

obelisk79, 2025-05-18:
The color assigned is part of an objects properties, and therefore saved to the FCStd file. My proposal would flag the default color as "default" and apply the users default color preference on load. If a user decides to assign specific colors to an object, the theme should not be modifying this color in any way (ie. Adjusting lightness/gamma) as it was an intentional act.

maxwxyz, 2025-05-18:
I'm against this proposal, as the file would look different on any system. Also, exporting the geometry (e.g. as STEP) would bake in the local color of the user. 
IMO best would be: the theme should **not** change the material.

maxwxyz, 2025-05-18:
Yes this is the issue: the default color for Part/PD objects which is also changed by the themes.

obelisk79, 2025-05-18:
The built in themes change the value? Or is this in reference to 3rd party add-ons?

pierreporte, 2025-05-18:
The new built-in light and dark themes do change the color. It’s not the same as with classic.

maxwxyz, 2025-05-18:
Yes, the default material is different in Classic, Dark and Light so the theme messes it up, also in 1.0

adrianinsaval, 2025-05-18:
can we blacklist some preferences that themes should not touch? maybe with an optionla toggle to allow chanigng them if the user really really wants that

maxwxyz, 2025-05-18:
IMO the preference should not exist as it overrides the default material

MisterMakerNL, 2025-05-18:
I tested if the contrast issue is still a thing since @kadet1090 told me the full white-face issue was fixed, and I think it isn't an issue anymore. So we can remove the setting from the theme preferences.