FreeCAD/FreeCAD #14993: App::PropertyEnumeration returns a string if addressed through an App::PropertyLink

krushia, 2024-06-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Most uses of an `App::PropertyEnumeration` in Expressions will return an integer (the index selected). When access is done through any variation of `App::PropertyLink`, the string is returned instead (bug).

I have found this occurs in every version of FreeCAD back to 0.21.2. I did not try any earlier versions.

A fix will likely break some existing projects relying on this to access the string, but it would be trivial to update such projects. Attached is a test case, showing some variations of access.

[enumeration-link-bug.FCStd.delete.this.extension.zip](https://github.com/user-attachments/files/16020152/enumeration-link-bug.FCStd.delete.this.extension.zip)

[image]

### Full version info

```shell
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37841 (Git) AppImage
Build type: Release
Branch: main
Hash: 5f761c758e2a0164b00a7183b9c0cbe32688e123
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * parts_library
  * Curves 0.6.40
  * A2plus 0.4.65 (Disabled)
  * DynamicData 2.62.0
  * Assembly4 0.50.13 (Disabled)
  * dodo 1.0.1
  * Alternate_OpenSCAD 1.0.0 (Disabled)
  * CurvedShapes 1.0.9
  * Design456 0.0.1
  * fasteners 0.5.24
  * SearchBar 1.0.2
  * Nodes 0.1.36 (Disabled)
  * 3D_Printing_Tools (Disabled)
  * workfeature (Disabled)
  * MnesarcoUtils 0.2.5
  * OpenTheme 2024.6.22
  * Manipulator 1.5.7
  * Assembly3 0.12.2 (Disabled)
  * lattice2 1.0.0
  * Launcher
  * CommandPanel
  * CubeMenu
  * Reinforcement
  * PieMenu 1.7.0
  * MeshRemodel 1.9.23
```

### Subproject(s) affected?

Expressions

### Anything else?

This bug is related to https://github.com/FreeCAD/FreeCAD/issues/14578

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

krushia, 2024-06-27:
Oops, there is a typo in rows 14 and 15. Please disregard those. Lines 6 and 7 accurately show the bug.

krushia, 2024-06-27:
Further testing reveals the bug goes beyond the scope in the current title (currently _App::PropertyEnumeration returns a string if addressed through an App::PropertyLink_).

1. This is not confined to `App::PropertyLink` (and derivative properties). In a spreadsheet, if C19 has `=<<foo>>._self`, putting `=.C19.TestEnumeration` in another cell will show the string instead of integer.

2. This is not confined to `App::PropertyEnumeration`. Weird things happen with any property type that has sub-properties. For example, referencing an `<<foo>>.TestPlacement.Rotation.Angle` will return degrees; but `=<<foo>>._self` <- `=.C19.TestPlacement.Rotation.Angle` will return radians.

I'd also like to clarify that this bug affects anything that takes an expression. It is not limited to spreadsheets.

krushia, 2024-06-28:
Python returns the string when accessing an `App::PropertyEnumeration` in all tested cases. This is returned as an actual `str` type, so the `.Enum` isn't accessible. This makes even less sense to me.

I'm really confused as to:
1. How `App::PropertyEnumeration` is **supposed** to work.
2. Why I can't find any other reports on this, which has been present for **at least** 3 years. In hindsight, this is the root cause of countless problems in my first FreeCAD project. I've done a multitude of redesign thinking I did something wrong. Are there really so few people who use expressions and links?

kaiwas, 2024-06-28:
> 2\. Are there really so few people who use expressions and links?
Perhaps this comment will bring some clarity.

https://github.com/FreeCAD/FreeCAD/issues/14670#issuecomment-2182959791

Perhaps there are really few people who use links. I'm just starting to master this topic. But I'm still very far from using the "code".

FEA-eng, 2025-07-17:
I get the following errors when opening this file:

```
13:26:32  <Spreadsheet> Cell.cpp(663): enumeration_link_bug#Spreadsheet.C15: No attribute named 'Enum'
13:26:32  <Spreadsheet> Cell.cpp(663): enumeration_link_bug#Spreadsheet.C14: No attribute named 'Enum'
13:26:32  <Spreadsheet> Sheet.cpp(1654): Failed to restore enumeration_link_bug#Spreadsheet: One or more cells failed contains errors.
```

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.42564 (Git)
Build date: 2025/07/14 18:51:42
Build type: Release
Branch: main
Hash: dc2f2374b6ab3e3381653fad125b1b619bb49f7c
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD_Stylesheet.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

FEA-eng, 2026-10-25:
Still relevant

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
OpenGL version/vendor/renderer: 4.6.0 - Build 27.20.100.9664/Intel/Intel(R) UHD Graphics 620
```