FreeCAD/FreeCAD #14998: Update View option- is the model supposed to update as we change it?

JDMils, 2024-06-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This issue has to do with the 3D model updating as properties are changed on the model. I created the following forum post to discuss the issue I am seeing and provided a sample model to test on. Forum discussion is here:
https://forum.freecad.org/viewtopic.php?p=767884#p767884

To further supplement the fact that this is a bug, after editing the 3D model's pad length and then deciding not to continue with the changes, I clicked Cancel which took me out of the Pad parameters dialog, the changes I had made were applied to the 3D model. And furthermore, no matter how many times I pressed CTRL-Z to undo the changes, they did not undo, so it seems that the change history was lost when Cancel was pressed however the changes were applied to the model.

Issue was reproduced in the following versions:
[code]
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Australia (en_AU)
[/code]

`OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37573 (Git)
Build type: Release
Branch: main
Hash: 91aac05021023c946dc1545f2b13512ccf326550
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
`

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Australia (en_AU)

And....
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37573 (Git)
Build type: Release
Branch: main
Hash: 91aac05021023c946dc1545f2b13512ccf326550
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Australia (en_AU)
```

### Subproject(s) affected?

None

### Anything else?

See forum post for model and full description.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

closed 2024-06-28 by maxwxyz

JDMils, 2024-06-29:
Hi, why is this issue closed? I have provided a sample model, a video to show the issue has been present since the 0.21 version up until the latest dev version yet it seems no one is interested in fixing this bug?

FlachyJoe, 2024-06-30:
@JDMils if nobody else can reproduce the issue we can't do anything.
Retry with the latest dev and if the problem is specific to a model please share it.

JDMils, 2024-06-30:
OK, I think I know what the problem is now. The model I am using is corrupt. When first loaded into FCAD, the model shows on screen however an error appears in the Report View:
Reading failed from embedded file: StringHasher.Table.txt (143 bytes, 80 bytes compressed)

I then double-click the Pad, the pad length is set to 25mm. I don't do anything except press OK and the model reduces the pad to 25mm on screen. I tested this in versions: 0.21, 37436, 37543, 37573 and 37928. So yes, I guess this is not a bug after all!
Thanks.