FreeCAD/FreeCAD #14999: Status bar toolbar names are not translatable

furgo16, 2024-06-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When right-clicking on the Status bar, the names of the toolbars are not translatable. They are also named as code variables instead of user-readable texts (e.g. `notificationArea` instead of `Notification area`)

[image]

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37915 (Git)
Build type: Release
Branch: main
Hash: 8d3f07c1f73ef5b1829760db251b9524094aeea2
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Installed mods: 
  * OpenTheme 2024.5.16
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-06-28:
I've given a go at fixing this. I got as far as making the "Notification area" and "Navigation indicator" translatable:

https://github.com/FreeCAD/FreeCAD/compare/main...furgo16:FreeCAD:status_area_toolbar_names

However, for  "draft snap widget" and "draft scale widget" are proving to be not at simple.

Syres916, 2024-06-28:
You could try something like https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:Draft_Status_Bar_Translate_Toolbar it's crucial that all entries in source are corrected.

furgo16, 2024-06-28:
Thank you, I'll give that a go.

I still wonder if there is a better way to find children widgets that does not rely on a user-visible string, so that the name can be decoupled from the search.

closed 2025-04-07 by chennes