FreeCAD/FreeCAD #15001: [Part] Sweep does not correctly pull elements with holes

kaiwas, 2024-06-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If we stretch an element that closes the holes, the holes open. And there is no way to open the holes.

https://github.com/FreeCAD/FreeCAD/assets/32273833/5a86da0a-5513-4fb4-9657-7098cf38a429

### Full version info

```shell
0,22
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FlachyJoe, 2024-06-28:
Note you can't select a sketch but a face with a hole.
If you add a hole in the sketch after you create the sweep, recompute fails with the message **Sweep: A fatal error occurred when making the sweep**

The problem is therefore to allow the selection of hollow faces.

kaiwas, 2024-06-28:
> Note you can't select a sketch

This is an important addition.
In my task, I planned to extrude a flat object (not a sketch), but probably someone may also need to extrude a sketch.

kaiwas, 2024-07-23:
I don't know where or when the necessary changes were made, but it looks like the sweep is now working as it should. 
[image]