FreeCAD/FreeCAD #15002: Assembly BOM: pre-populate Description column with Label2 contents

furgo16, 2024-06-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

For a more automated user experience, it would be optimal for the generated Bill Of Materials to pre-populate the default `Description` column with the contents of the components' `Label2` property, if there is one.

This suggestion is only for the initial population of the column, the rest (Description not being overwritten upon recalculation) would stay the same.

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37915 (Git)
Build type: Release
Branch: main
Hash: 8d3f07c1f73ef5b1829760db251b9524094aeea2
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Installed mods: 
  * OpenTheme 2024.5.16
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-07-01:
Imo description should be a property string of the document object.
Also document objects should have a part number property.
@pierreporte has opened an issue about this.

furgo16, 2024-07-01:
I couldn't find that issue, but if I understand it correctly, having a document-wide description would only work for assemblies where there is only a part per file. When all parts are in the same file, a part-specific property would be needed. The suggestion of `Label2` was to support something that's already available (and often used as a description) for a first working iteration.

pierreporte, 2024-07-01:
@furgo16 The issue mentioned by @PaddleStroke is likely #12136.

furgo16, 2025-07-17:
This is still valid, but less of an issue, as with https://github.com/FreeCAD/FreeCAD/pull/20732 there is now a workaround. 

This PR is also somewhat related: https://github.com/FreeCAD/FreeCAD/pull/16990